openhwfoundation/cva6 #541: DTM issue during negotiation time when XLEN=32bits

JeanRochCoulon, 2020-10-06:
@zarubaf

I come back to you on a DTM subject, already discussed 1 month ago. My feeling is that DTM must be modified to support XLEN=32. 

When debug is activated and XLEN=32, I get an exception. During the negotiation phase, the SimDTM module challenges the DTM to know which is the size of the microprocessor. First SimDTM command tries to ask for 128bits transaction, second command asks for 64bits. If the two commands are not supported, SimDTM understands it is 32bits.

The first debug command received by DTM is: 0x00421008, size=3’b100 (128bits)
  The DTM answers with  
0x0000000000000300 ebreak 
 because the 128bits is not supported

The second command received is : 0x00321008, size=3’b011 (64bits)
  The DTM answer with the following sequence of instructions
0x0000000000000300  j       pc + 0x38
0x0000000000000338  csrw    unknown_7b3, a0
0x000000000000033c  auipc   a0, 0x0
0x0000000000000340  srli    a0, a0, 12
0x0000000000000344  slli    a0, a0, 12
0x0000000000000348  sd      s0, 896(a0)
As sd is not implemented in rv32, exception occurs

To my mind the second command should not be supported by DTM and DTM should be modified. I would be please to know you feeling about it. In the case you confirm it, I could submit a PR.

zarubaf, 2020-10-07:
I am pretty confident the debug module should not be changed :-) In fact, we are using it for all our 32-bit cores and lowrisc does it as well with their Ibex core. If the exception occurs the debug module should detect that it does not support the `sd` and hence interacts with a 32-bit core.

I am not sure that I understand the problem you are facing. Can you elaborate?

JeanRochCoulon, 2020-10-07:
Lowrisc and Ibex are 32bits CPU with 32bits bus width. That is not the case for Ariane 32bits which get 64bit bus width.

The SimDTM module executes 2 challenge commands to determine the RV ISA (RV128, RV64 or RV32) to be supported:
- First, SimDTM executes the 128bit command,
- If the HW DTM supports the 128bit command, HW DTM is set in RV128 configuration, else SimDTM executes the 64bit command,
- If the HW DTM supports the 64bit command, HW DTM is set in RV64 configuration, else HW DTM is set in RV32 configuration 

The issue is that HW DTM looks at the bus width, and not the XLEN, to determine if the command is supported or not. The bug is not revealed by lowriscv or Ibex because busWidth=XLEN. In Ariane32 case, the HW DTM takes the decision to be RV64 by evaluating the bus width (64bits), which is wrong!

I hope this explanation will clarify the issue. If not, we can plan a short call. Cheers

JeanRochCoulon, 2020-10-07:
In another words, DTM cannot support CPU with bus width different from core data width. Indeed, buswidth is an input parameter of dm_top.sv but there is no input parameter for the core data width!

zarubaf, 2023-10-20:
I think we fixed this, didn't we @JeanRochCoulon?

closed 2023-10-20 by zarubaf

JeanRochCoulon, 2023-10-20:
Indeed !!!!!!!!!