openhwfoundation/cva6 #542: Cache Subsystem Doc

lewiz-kiratey, 2020-10-07:
Hello,
Is there any documentation on the OpenPiton compatible 'wt_cache_subsystem' that describes the structure and functioning of the subsystem?
I am trying to understand the D$ which has 2-way SRAM for data and 4-way SRAM for tag?
Any documentation or lead is helpful.
Thanks

epthales, 2020-10-27:
Hello, 
I'm a trainee engineer at Thales France and I did a little documentation of the CVA6 cache subsystem. If you're interested I can send you the documentation by mail.
Regards,
Emeric

gabaxter, 2020-12-11:
I have the same question and would love to see if there is an update to this issue.

bufans, 2021-05-28:
same question too. Any cache subsystem microarchitecture diagram ?

fatimasaleem, 2021-12-29:
I have the same question. @lewiz-kiratey @gabaxter @bufans Did any of you find any helpful piece of documentation on wt_cache_subsystem? cc @msfschaffner @zarubaf

MikeOpenHWGroup, 2023-02-17:
Hello @kirateys, @gabaxter, @bufans and @fatimasaleem.  Thanks for your interest in CVA6.

The CVA6 project has started to focus on updating the User Manual and Design documents.  It is published by ReadTheDocs ([link](https://docs.openhwgroup.org/projects/cva6-user-manual/)).  This is still very much  a work-in-progress, but I expect you'll be happy with the result.  Please check back regularily for updates, and if you have comments and questions, please generate an issue against the documentation itself.  Better yet, you can always make a contribution!

In the meantime, I will close this specific issue since it has not had an update for more than one year.

closed 2023-02-17 by MikeOpenHWGroup