openhwfoundation/cva6 #543: failed to dump vcd

rudyyao, 2020-10-10:
Hi there,

I'm trying to dump vcd file using below command, but the 'rv64ui-p-add.vcd' file's size is 0 when the test finished.
Can anyone help on this?

work-ver/Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add

I have run: make verilate DEBUG=1 before running the above command.
Also the hello world test can run correctly.

FranDuqueAyachi, 2022-06-04:
> Hi there,
>
> I'm trying to dump vcd file using below command, but the 'rv64ui-p-add.vcd' file's size is 0 when the test finished. Can anyone help on this?
>
> work-ver/Variane_testharness -v [...]

try to use this path instead the one you use:
work-ver/Variane_testharness -v rv64ui-p-add.vcd tmp/riscv-tests/build/isa//rv64ui-p-add

MikeOpenHWGroup, 2023-02-17:
Thanks for your help with this @FranDuqueAyachi .

Hi @rudyyao , thanks for your interest in CVA6. This issue has not been updated in more than nine months (and you have not commented for more than two years), so I will assume it is resolved and will close this issue. If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup