openhwfoundation/cva6 #545: working with VC709 FPGA or any virtex 7 family

gfsdllc, 2020-10-13:
what ae the files which needed to be able to make ariane support VC709 or VC7020 or any virtex 7 family??

xushengj, 2020-12-25:
I'm not aware of any Xilinx FPGA board called "VC7020". We have a working VC709 port (not publicly available yet), but it is not very usable. The main issue with VC709 is that (1) the built-in ethernet module does not work on VC709; (2) VC709 does not have an SD card reader. Our VC709 port ended up using the BPI flash to store the SD card content along with the bitstream, and we can only use the serial port to transfer files (with base64). If this is still okay with you then please let me know and we may create a pull request that you can use. CVA6 supports VC707, and if possible I recommend to get one VC707.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insight @xushengj .

Hi @gfsdllc, thanks for your interest in CVA6. This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue. If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup