openhwfoundation/cva6 #546: Can we Run RTOS?

suhasskrishanmurthy, 2020-10-14:
Hi,
    I am trying to run Freertos on ariane.
   I have two tasks task1 and task2, task1 will print something and enters block state there by giving time for task2 to print something and control goes back to task1 and process continues.
  while running on questa task1 will print and give control to task2 and then task2 will print but control is not going back to task1.
 I have ran the same elf on SPIKE and its working fine.
Why this is happening any idea?
Thanks in advance.

zarubaf, 2020-10-14:
So Ariane has all the privileged specification implemented, the most of it you won't even need as far as I remember FreeRTOS. Unfortunately, I think with the high-level problem description I can't really help you.

MikeOpenHWGroup, 2023-02-17:
Hi @suhasskrishanmurthy,  thanks for your interest in CVA6.  This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup