openhwfoundation/cva6 #548: MMU Performance Issue

jctullos, 2020-10-17:
I wanted to see if there have been any findings on the MMU for Ariane in regards to slower than normal operation. It's hard to characterize what normal operation is, but I have 2 other RISC-V processors with PMP that conduct some benchmarks along with Ariane. 

All 3 are testing the page table walker and verifying the virtual and physical memory. The other 2 are running at around the same cycle count within + or - 1 second within each other (I'm leaving out the actual cycle count for brevity). For Ariane, it's taking an additional 60 seconds (at 50MHz) on top of the time it takes the other 2 processors to do the same code for the PTW/TLB. I'm not entirely sure how to go about debugging something like this, but thought I would bring it up in case someone else has seen similar findings.

zarubaf, 2020-10-19:
That is honestly a bit strange. I would consider the PTW walking close to optimal, in any case, that shouldn't be the case too often as you have TLBs preventing redundant walks. 
Which cache configuration do you use (write-through or write-back)? If you are using the write-back cache the `fence` instruction takes quite some time writing back and invalidating its entries. 
Potentially one thing which could also cause problems is the treatment of the global bit which is basically untested in Ariane as far as I know/remember, there might be a performance bug lingering.
Would you be able to share snippets of the code which uncover that problem?

jctullos, 2020-10-19:
Could I email you some snippets?

I figured there might be something else going on. I'm using write-through cache, so I wouldn't expect to have any issues there.

zarubaf, 2020-10-19:
Yes absolutely, please!

MikeOpenHWGroup, 2023-02-17:
Hi @zarubaf. This issue has not been updated in more than two years, but it does not appear to be resolved.  What is the status of this issue?

zarubaf, 2023-10-20:
I don't recall whether there was any outcome. Closing because date opened.

closed 2023-10-20 by zarubaf