openhwfoundation/cva6 #553: errors: about "make verilate DEBUG=1"

ckj119940887, 2020-11-06:
When I executed "make verilate DEBUG=1", I encounterd the following errors:

../tb/ariane_tb.cpp:113:10: error: ‘bool preload_aware_dtm_t::is_address_preloaded(addr_t, size_t)’ marked ‘override’, but does not override
     bool is_address_preloaded(addr_t taddr, size_t len) override { return true; }

my verilator version is : Verilator 4.008

jatanpandya83, 2021-01-28:
I am also seeing this issue? Is this resolved? Any  suggestions will be helpful!

Moschn, 2021-09-28:
For a quick fix, you can just remove the `override` keyword in the line mentioned in the error message.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insight @Moschn .

Hi @ckj119940887 , thanks for your interest in CVA6. This issue has not been updated in more than a year, so I will assume it is resolved and will close this issue. Note that I typically run using Verilator v4.100.

If you are still having trouble, please feel free to open another issue.

closed 2023-02-17 by MikeOpenHWGroup