openhwfoundation/cva6 #555: kernel hanging randomly

monniaux, 2020-11-12:
I'm running rev `a7ad3ec` compiled with Vivado 2020.1.1 for running on a Genesys 2 board.
```
$ md5sum bbl.bin 
632f22d7ff2287ab1d41aa87bbd4656f  bbl.bin
```
When I run programs, sometimes the kernel hangs. Sometimes it shows this kind of messages
```
[  972.257681] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  972.257681] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  972.269937] rcu:     (detected by 0, t=14712 jiffies, g=4733, q=0)
[  972.269937] rcu:     (detected by 0, t=14712 jiffies, g=4733, q=0)
[  972.290499] rcu: All QSes seen, last rcu_sched kthread activity 14712 (4295034519-4295019807), jiffies_till_next_fqs=1, root ->qsmask 0x0
[  972.290499] rcu: All QSes seen, last rcu_sched kthread activity 14712 (4295034519-4295019807), jiffies_till_next_fqs=1, root ->qsmask 0x0
[  972.331864] bash            R  running task        0   149    121 0x00000008
[  972.331864] bash            R  running task        0   149    121 0x00000008
[  972.356088] Call Trace:
[  972.356088] Call Trace:
[  972.370822] [<ffffffe000693fd0>] walk_stackframe+0x0/0xa0
[  972.370822] [<ffffffe000693fd0>] walk_stackframe+0x0/0xa0
[  972.384894] [<ffffffe0006941cc>] show_stack+0x2a/0x34
[  972.384894] [<ffffffe0006941cc>] show_stack+0x2a/0x34
[  972.402775] [<ffffffe0006b13ec>] sched_show_task+0xb0/0x10a
[  972.402775] [<ffffffe0006b13ec>] sched_show_task+0xb0/0x10a
[  972.421220] [<ffffffe0006d6fc2>] rcu_sched_clock_irq+0x4f2/0x4fa
[  972.421220] [<ffffffe0006d6fc2>] rcu_sched_clock_irq+0x4f2/0x4fa
[  972.441207] [<ffffffe0006db4a0>] update_process_times+0x22/0x4c
[  972.441207] [<ffffffe0006db4a0>] update_process_times+0x22/0x4c
[  972.461675] [<ffffffe0006e432e>] tick_periodic+0x44/0xb0
[  972.461675] [<ffffffe0006e432e>] tick_periodic+0x44/0xb0
[  972.480854] [<ffffffe0006e43b8>] tick_handle_periodic+0x1e/0x5c
[  972.480854] [<ffffffe0006e43b8>] tick_handle_periodic+0x1e/0x5c
[  972.500453] [<ffffffe0008fa0f6>] riscv_timer_interrupt+0x32/0x3a
```

What can I do about this ?

Steinegger, 2020-11-20:
We regularly encountered this issue while doing benchmarks as well. Sometimes it results in a kernel panic like you described
<details>
  <summary>Log output</summary>

```
[  319.968454] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  319.968454] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  319.975861] rcu:     (detected by 0, t=2102 jiffies, g=6985, q=2)
[  319.975861] rcu:     (detected by 0, t=2102 jiffies, g=6985, q=2)
[  319.988096] rcu: All QSes seen, last rcu_sched kthread activity 2102 (4294969284-4294967182), jiffies_till_next_fqs=1, root ->qsmask 0x0
[  319.988096] rcu: All QSes seen, last rcu_sched kthread activity 2102 (4294969284-4294967182), jiffies_till_next_fqs=1, root ->qsmask 0x0
[  320.012673] mount           R  running task        0   118    108 0x00000008
[  320.012673] mount           R  running task        0   118    108 0x00000008
[  320.027186] Call Trace:
[  320.027186] Call Trace:
[  320.036082] [<ffffffe0012b7ff4>] walk_stackframe+0x0/0xa0
[  320.036082] [<ffffffe0012b7ff4>] walk_stackframe+0x0/0xa0
[  320.044485] [<ffffffe0012b81f0>] show_stack+0x2a/0x34
[  320.044485] [<ffffffe0012b81f0>] show_stack+0x2a/0x34
[  320.055224] [<ffffffe0012d541a>] sched_show_task+0xb0/0x10a
[  320.055224] [<ffffffe0012d541a>] sched_show_task+0xb0/0x10a
[  320.066300] [<ffffffe0012faff0>] rcu_sched_clock_irq+0x4f2/0x4fa
[  320.066300] [<ffffffe0012faff0>] rcu_sched_clock_irq+0x4f2/0x4fa
[  320.078291] [<ffffffe0012ff4ce>] update_process_times+0x22/0x4c
[  320.078291] [<ffffffe0012ff4ce>] update_process_times+0x22/0x4c
[  320.090576] [<ffffffe00130835c>] tick_periodic+0x44/0xb0
[  320.090576] [<ffffffe00130835c>] tick_periodic+0x44/0xb0
[  320.102086] [<ffffffe0013083e6>] tick_handle_periodic+0x1e/0x5c
[  320.102086] [<ffffffe0013083e6>] tick_handle_periodic+0x1e/0x5c
[  320.113833] [<ffffffe00151f646>] riscv_timer_interrupt+0x32/0x3a
[  320.113833] [<ffffffe00151f646>] riscv_timer_interrupt+0x32/0x3a
[  320.126118] [<ffffffe00165622a>] do_IRQ+0xaa/0xb6
[  320.126118] [<ffffffe00165622a>] do_IRQ+0xaa/0xb6
[  320.137017] [<ffffffe0012b6f98>] ret_from_exception+0x0/0xc
[  320.137017] [<ffffffe0012b6f98>] ret_from_exception+0x0/0xc
[  320.147758] rcu: rcu_sched kthread starved for 2102 jiffies! g6985 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=0
[  320.147758] rcu: rcu_sched kthread starved for 2102 jiffies! g6985 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=0
[  320.167408] rcu: RCU grace-period kthread stack dump:
[  320.167408] rcu: RCU grace-period kthread stack dump:
[  320.179135] rcu_sched       R  running task        0    10      2 0x00000000
[  320.179135] rcu_sched       R  running task        0    10      2 0x00000000
[  320.192394] Call Trace:
[  320.192394] Call Trace:
[  320.201198] [<ffffffe001653504>] __schedule+0x1b4/0x4ea
[  320.201198] [<ffffffe001653504>] __schedule+0x1b4/0x4ea
[  320.209477] [<ffffffe001653854>] schedule+0x1a/0x4e
[  320.209477] [<ffffffe001653854>] schedule+0x1a/0x4e
[  320.219869] [<ffffffe00165579c>] schedule_timeout+0xe8/0x142
[  320.219869] [<ffffffe00165579c>] schedule_timeout+0xe8/0x142
[  320.230869] [<ffffffe0012fa1d0>] rcu_gp_kthread+0x464/0x846
[  320.230869] [<ffffffe0012fa1d0>] rcu_gp_kthread+0x464/0x846
[  320.242484] [<ffffffe0012d126e>] kthread+0xce/0xdc
[  320.242484] [<ffffffe0012d126e>] kthread+0xce/0xdc
[  320.253062] [<ffffffe0012b6f98>] ret_from_exception+0x0/0xc
[  320.253062] [<ffffffe0012b6f98>] ret_from_exception+0x0/0xc

```

</details>

... and sometimes the device would just freeze and not react to any inputs. Might this be connected to #517, #476 or #480?
It makes benchmarking, especially longer running benchmarks really hard to do, because booting takes quite a significant amount of time and regularly checking if it's still "alive" is rather tedious.

monniaux, 2021-01-21:
version eaeef7d
Vivado 2019.2.1

[  916.125026] [<ffffffe0006941cc>] show_stack+0x2a/0x34
[  916.125026] [<ffffffe0006941cc>] show_stack+0x2a/0x34
[  916.142891] [<ffffffe0006b13ec>] sched_show_task+0xb0/0x10a
[  916.142891] [<ffffffe0006b13ec>] sched_show_task+0xb0/0x10a
[  916.161365] [<ffffffe0006d6fc2>] rcu_sched_clock_irq+0x4f2/0x4fa
[  916.161365] [<ffffffe0006d6fc2>] rcu_sched_clock_irq+0x4f2/0x4fa
[  916.181317] [<ffffffe0006db4a0>] update_process_times+0x22/0x4c
[  916.181317] [<ffffffe0006db4a0>] update_process_times+0x22/0x4c
[  916.201804] [<ffffffe0006e432e>] tick_periodic+0x44/0xb0
[  916.201804] [<ffffffe0006e432e>] tick_periodic+0x44/0xb0
[  916.220945] [<ffffffe0006e43b8>] tick_handle_periodic+0x1e/0x5c
[  916.220945] [<ffffffe0006e43b8>] tick_handle_periodic+0x1e/0x5c
[  916.240558] [<ffffffe0008fa0f6>] riscv_timer_interrupt+0x32/0x3a
[  916.240558] [<ffffffe0008fa0f6>] riscv_timer_interrupt+0x32/0x3a
[  916.261013] [<ffffffe000a30ca2>] do_IRQ+0xaa/0xb6
[  916.261013] [<ffffffe000a30ca2>] do_IRQ+0xaa/0xb6
[  916.279190] [<ffffffe000692f74>] ret_from_exception+0x0/0xc
[  916.279190] [<ffffffe000692f74>] ret_from_exception+0x0/0xc
[  916.297054] [<ffffffe0006f136c>] find_get_entry+0x4e/0x102
[  916.297054] [<ffffffe0006f136c>] find_get_entry+0x4e/0x102
[  916.316045] rcu: rcu_sched kthread starved for 8407 jiffies! g15021 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=0
[  916.316045] rcu: rcu_sched kthread starved for 8407 jiffies! g15021 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=0
[  916.349026] rcu: RCU grace-period kthread stack dump:
[  916.349026] rcu: RCU grace-period kthread stack dump:
[  916.368551] rcu_sched       R  running task        0    10      2 0x00000000
[  916.368551] rcu_sched       R  running task        0    10      2 0x00000000
[  916.390637] Call Trace:
[  916.390637] Call Trace:
[  916.405317] [<ffffffe000a2df7c>] __schedule+0x1b4/0x4ea
[  916.405317] [<ffffffe000a2df7c>] __schedule+0x1b4/0x4ea
[  916.419100] [<ffffffe000a2e2cc>] schedule+0x1a/0x4e
[  916.419100] [<ffffffe000a2e2cc>] schedule+0x1a/0x4e
[  916.436431] [<ffffffe000a30214>] schedule_timeout+0xe8/0x142
[  916.436431] [<ffffffe000a30214>] schedule_timeout+0xe8/0x142
[  916.454760] [<ffffffe0006d61a2>] rcu_gp_kthread+0x464/0x846
[  916.454760] [<ffffffe0006d61a2>] rcu_gp_kthread+0x464/0x846
[  916.474110] [<ffffffe0006ad240>] kthread+0xce/0xdc
[  916.474110] [<ffffffe0006ad240>] kthread+0xce/0xdc
[  916.491735] [<ffffffe000692f74>] ret_from_exception+0x0/0xc
[  916.491735] [<ffffffe000692f74>] ret_from_exception+0x0/0xc

Steinegger, 2021-03-03:
I wanted to add a few more details to this issue:
We've encountered it on both the Genesys2 board and the KC705. In both case we encountered it with the stock pre-built ariane-sdk image from the releases page and self-built linux images.

xushengj, 2021-03-07:
Hi all, we also encountered this issue. We resolved this by setting
`CONFIG_SMP=n`
In the Linux configuration. Hope this works for you.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insight @xushengj and @Steinegger .

Hi @monniaux, thanks for your interest in CVA6. This issue has not been updated in almost two years, so I will assume it is resolved and will close this issue. If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup