openhwfoundation/cva6 #558: How does CVA6 handle AXI response errors?

Silabs-ArjanB, 2020-11-17:
Hi,

I am trying to figure out how CVA6 is handling response errors on its AXI interfaces, but I cannot find it in the documentation. My specific question is how responses other than RESP_OKAY are dealt with. Will they lead to precise or imprecise exceptions?

zarubaf, 2020-11-17:
Unfortunately, they are not handled at the moment. They will likely cause troubles as the value does not make sense in most of the cases, but that can happen many cycles downstream in the program.

If they were to be handled then they would likely be imprecise as the store buffer lives pretty much outside the pipeline. Loads could be handled precisely. Does that make sense?

Silabs-ArjanB, 2020-11-17:
Hi @zarubaf Thank you for the quick answer. Why do you expect that loads can be handled precisely; do you assume that loads will never cause side effects and would you somehow guarantee that a store which follows a load in program order never makes it onto the AXI bus until the load has signaled whether it is okay or not?

Lemover, 2023-09-13:
I guess: 
* load's precise axi bus error is easy to implement. Because the load instr will wait until the bus' response is back. 
* But store instruction will commit and store the data into store commit buffer. When the data at the store commit buffer is stored into cache hierarchy, the bus error is raised. So store bus error is easy to implement imprecisely.

zarubaf, 2023-10-20:
AXI error responses are not handled see issue #2

closed 2023-10-20 by zarubaf