openhwfoundation/cva6 #563: main.o: can't link soft-float modules with double-float modules

yangjingchi, 2020-11-28:
i got the following error when try to cross-compile a hello world program with printf.
I google the error and it seems that it was a know issue for [gnu-toolchain](https://lists.gnu.org/archive/html/bug-binutils/2019-03/msg00184.html) but i think it's fixed few month ago 

Also i notice that in the fpga bootrom, there is a uart_printf in main function, Is there a reason that bare metal program not using the standard printf?

riscv64-unknown-elf-gcc -Os -march=rv64imac -mabi=lp64 -Wall -mcmodel=medany -mexplicit-relocs -I./ -c main.c  -o main.o
main.c  ==>  main.o

### **_riscv64-unknown-elf-gcc -Os -march=rv64imac -mabi=lp64 -Wall -mcmodel=medany -mexplicit-relocs  -I./ -Tlinker.lds main.o -o main.elf_**
**/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: main.o: can't link soft-float modules with double-float modules**
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file main.o
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/crt0.o: in function `_start':
(.text+0x0): undefined reference to `__global_pointer$'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/bin/ld: (.text+0x10): undefined reference to `_end'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/crtbegin.o: in function `__do_global_dtors_aux':
crtstuff.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against `completed.1'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): in function `exit':
exit.c:(.text+0xe): relocation truncated to fit: R_RISCV_HI20 against symbol `_global_impure_ptr' defined in .srodata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-printf.o): in function `_printf_r':
printf.c:(.text+0x22): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfprintf.o): in function `.L3':
vfprintf.c:(.text+0xac): relocation truncated to fit: R_RISCV_HI20 against `.LC10'
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-wsetup.o): in function `__swsetup_r':
wsetup.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): in function `__call_exitprocs':
__call_atexit.c:(.text+0x2): relocation truncated to fit: R_RISCV_HI20 against symbol `_global_impure_ptr' defined in .srodata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): in function `register_fini':
__call_atexit.c:(.text.startup+0x6): relocation truncated to fit: R_RISCV_HI20 against symbol `__libc_fini_array' defined in .text section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o): in function `.L60':
fflush.c:(.text+0x1cc): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): in function `_cleanup_r':
findfp.c:(.text+0x4): relocation truncated to fit: R_RISCV_HI20 against symbol `_fclose_r' defined in .text section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fclose.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-freer.o): in function `_malloc_trim_r':
mallocr.c:(.text+0x4): relocation truncated to fit: R_RISCV_HI20 against symbol `__malloc_av_' defined in .data section in /opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/10.1.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-ldtoa.o): in function `.L272':
ldtoa.c:(.text+0xa24): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
make: *** [main.elf] Error 1

Chakri-myneni, 2020-12-01:
[https://github.com/riscv/riscv-gnu-toolchain/issues/784](url)
@yangjingchi  please refer to the above link.
I think the problem is related to version of your gcc.

zarubaf, 2020-12-14:
You need to have a `multilib` toolchain installed. You can also get a pre-built toolchain [from SiFive](https://www.sifive.com/software) (which is what I would recommend).

closed 2020-12-14 by zarubaf