openhwfoundation/cva6 #566: error occur during compiling verilator.h

S-Nomii, 2020-12-01:
Hi, I am trying to build simulation by make verilate, but it tells me that in tb/dpi/verilator.h file VerilatedVcdFile can not be derived. Then I find out the VerilatedVcdFile is a final type class in verilator include files. Can anyone tell me how it works.

zarubaf, 2020-12-14:
It seems that somehow your Verilator installation is broken. Do you have your `$VERILATOR_ROOT` set correctly?

TarekIbnZiad, 2021-01-17:
Hello,

I encountered the same error on ARCH linux with `verilator 4.106`. It works fine when I switched to the latest verilator from their repo. ` Verilator 4.109 devel rev v4.108-18-g5cadabeb`.

Thank you!

zarubaf, 2021-01-18:
Just tried the same. My advice, for now, is to stick to `4.104`.

Unfortunately, ` 4.108` or `4.109` have different problems (#585). The Verilator simulation has been stable for some time but it seems that this has changed with the latest two releases.

closed 2021-01-18 by zarubaf