openhwfoundation/cva6 #567: Installation of cva6

SubhraKantiDas2020, 2020-12-04:
Hi. I am trying to install cva6 on my Ubuntu 18.04 machine. I have successfully installed Riscv tools and gnu toolchain. When I try to "make verilate", I get a makefile error saying "recipe for target verilate failed". I have cloned the repo from master branch. Need help on this.

gabaxter, 2020-12-10:
Hello,

I suspect something wrong with the version of verilator that you have installed.  I did a simple `sudo apt -y install verilator`. For me this put Verilator 3.916 on 18.04 machine.  I got the same error as above albeit with an unsupported assert in a particular file.   

I then chose to build verilator from git source.  After a reboot I got much farther running `make verilate`.  Now I get a unsupported static cast.  This is with Verilator 4.107.

**Here are details**
```
user@machine:~/dev/cva6(master)$ verilator --version
Verilator 4.107 devel rev v4.106-55-g9a32158b

```

The two "new" errors which cause `make verilate` to fail after some time are:
```
%Error-UNSUPPORTED: /home/glenn/dev/cva6/tb/ariane_testharness.sv:566:20: Unsupported: static cast to 'logic[63:0][9:0][0:0]' from 'logic[639:0]'
                                                                        : ... In instance ariane_testharness
  566 |     .start_addr_i (addr_map_t'({
      |                    ^~~~~~~~~~
%Error-UNSUPPORTED: /home/glenn/dev/cva6/tb/ariane_testharness.sv:578:20: Unsupported: static cast to 'logic[63:0][9:0][0:0]' from 'logic[639:0]'
                                                                        : ... In instance ariane_testharness
  578 |     .end_addr_i   (addr_map_t'({
      |                    ^~~~~~~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual and https://verilator.org for more assistance.
Makefile:478: recipe for target 'verilate' failed
make: *** [verilate] Error 1
```

There appears to be some issues with versioning.  So can someone please tell us what versions of verilator work for you with the present release of cva6 (master branch).  Or in the alternate, what branch we should be using to build all of cva6?

Thank you!

Glenn

zarubaf, 2020-12-14:
This version for sure works for me:
```
% verilator --version                                                                                                                                     
Verilator 4.035 devel rev v4.034-100-g01f3e81a
```

Nevertheless, I would appreciate it if it would also run with the newest version. Can you try removing the `addr_map_t` casts and see whether it compiles? If yes, could you please submit a PR?

TarekIbnZiad, 2021-01-17:
Hello, 

I encountered the same error with `Verilator 4.109 devel rev v4.108-18-g5cadabeb`. I confirm that removing the `addr_map_t` casts resolves the compilation problem.    

Thank you,
Mohamed

commit d4605e3b (2021-01-18): verilator: Remove static casts fixes #567 (#582)

Static casts cause a compilation error with Verilator 4.109
closed 2021-01-18 by zarubaf