openhwfoundation/cva6 #570: Runtime error in VCS

abdullahyildiz, 2020-12-11:
I'm trying to build the simulation executable in VCS. Here is the respective target that I added to Makefile:

```
vcs_sim: $(dpi-library)/ariane_dpi.so
        mkdir -p work
        vlogan -full64 -q -ntb_opts uvm-1.1 -sverilog
        mkdir -p work/.build-srcs
        vlogan -full64 -nc -sverilog +define+WT_DCACHE $(filter %.sv,$(ariane_pkg)) +incdir+src/common_cells/include/
        vlogan -full64 -nc -sverilog -ntb_opts uvm-1.1 +define+WT_DCACHE $(filter %.sv,$(util)) +incdir+src/common_cells/include/+src/util/
        vhdlan -full64 $(filter %.vhd,$(uart_src))
        vlogan -full64 -nc -sverilog -assert svaext +define+WT_DCACHE $(filter %.sv,$(src)) +incdir+src/common_cells/include/+src/util/
        touch $(library)/.build-srcs
        vlogan -full64 -nc -sverilog -ntb_opts uvm-1.1 $(tbs)
        touch $(library)/.build-tb
        vcs -full64 work.ariane_tb
```

I get no error during compilation. However, when I run simv, it gives the following error:

```
$ ./simv -sv_lib work-dpi/ariane_dpi +UVM_TESTNAME=core_test
Chronologic VCS simulator copyright 1991-2019
Contains Synopsys proprietary information.
Compiler version P-2019.06_Full64; Runtime version P-2019.06_Full64;  Dec 11 13:20 2020

Error-[DPI-DIFNF] DPI import function not found
/eda/synopsys/vcs/P-2019.06/etc/uvm-1.1/base/uvm_resource.svh, 390
  The definition of DPI import function/task 'uvm_glob_to_re' does not exist.
  Please check the stated DPI import function/task is defined, and its 
  definition is either passed in a source file at compile-time, or provided in
  a shared library specified using the LRM Annex-J options at run-time.

$finish at simulation time                    0
           V C S   S i m u l a t i o n   R e p o r t 
Time: 0 ns
CPU Time:      0.490 seconds;       Data structure size: 513.5Mb
Fri Dec 11 13:20:17 2020
```

Is there anything missing during compilation?

I've attached the run log.
[run.log](https://github.com/openhwgroup/cva6/files/5678632/run.log)

zarubaf, 2020-12-14:
Looks like your installation has some problem with the UVM part. My understanding is, that the UVM DPI is part of the VCS installation. Seems to be more a problem of the VCS installation than the CVA6 source. Maybe try another version of UVM?

abdullahyildiz, 2020-12-14:
@zarubaf,

Thanks for the answer.

I get the same error both with UVM 1.1 and UVM 1.2, which are included in Synopsys VCS P-2019.06. Which version of VCS have you tried before?

zarubaf, 2020-12-14:
We usually use Questasim, so I can't really answer that question :/ I know our friends at Princeton use VCS, maybe they can share which version works for them (but I am pretty confident that the newest VCS version should work). Are you sure, your installation is correct? The tool doesn't seem to be able to find some internal UVM files. Maybe there is a UVM related command-line argument missing in your call to `simv`?

abdullahyildiz, 2020-12-14:
Here is my environment settings:

export VCS_UVM_HOME=/eda/synopsys/vcs/P-2019.06/etc/uvm-1.2
export VCS_HOME=/eda/synopsys/vcs/P-2019.06
export PATH=$PATH:$VCS_HOME/bin

I will also check with a newer version of VCS and share the results.

abdullahyildiz, 2020-12-14:
@zarubaf,

I still get the same error in VCS Q-2020.03-SP2. Possibly I'm doing something wrong during compilation or run-time.

zhuazizhua, 2020-12-18:
@abdullahyildiz,
Hello, I am running with a similar problem to yours, but I hope I can solve it by executing "vlogan -full64 -nc -sverilog +$(VCS_HOME)/etc/uvm-1.2/macros"
But I have the following problems:
`[bzhang@eda cva6]$ ./simv -sv_lib work-dpi/ariane_dpi +UVM_TESTNAME=core_test
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Dec 18 11:32 2020
----------------------------------------------------------------
UVM-1.1d.Synopsys
(C) 2007-2013 Mentor Graphics Corporation
(C) 2007-2013 Cadence Design Systems, Inc.
(C) 2006-2013 Synopsys, Inc.
(C) 2011-2013 Cypress Semiconductor Corp.
----------------------------------------------------------------

  ***********       IMPORTANT RELEASE NOTES         ************

  You are using a version of the UVM library that has been compiled
  with `UVM_NO_DEPRECATED undefined.
  See http://www.eda.org/svdb/view.php?id=3313 for more details.

  You are using a version of the UVM library that has been compiled
  with `UVM_OBJECT_MUST_HAVE_CONSTRUCTOR undefined.
  See http://www.eda.org/svdb/view.php?id=3770 for more details.

      (Specify +UVM_NO_RELNOTES to turn off this notice)

[TRACER] Output filename is: trace_hart_0.log
htif: invalid option -- 's'
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Unknown argument (did you mean to enable +permissive parsing?)`

zhuazizhua, 2020-12-18:
@abdullahyildiz，
Hello, I have encountered similar errors, I hope my solution is helpful to you, in addition to the file is available:“vlogan -full64 -nc -sverilog+$(VCS_HOME)/etc/uvm-1.2/macros”

zhuazizhua, 2020-12-18:
Hello, I have encountered similar errors, I hope my solution is helpful to you, in addition to the file is available:"vlogan -full64 -sverilog +v2k -f pkg.f +incdir+$(VCS_HOME)/etc/uvm-1.2/macros".You can find the missing "UVM *" in the $(VCS_HOME) directory by "find".My English is a little weak. You have to try to understand

abdullahyildiz, 2020-12-18:
@zhuazizhua,

Thanks for help. I'll try and let you know.

abdullahyildiz, 2020-12-18:
@zhuazizhua,

As far as I understand, you ask me to add the command `vlogan -full64 -sverilog +v2k -f pkg.f +incdir+$(VCS_HOME)/etc/uvm-1.2/macros` to the list of commands that I executed.

I added the command above to the Makefile but I didn't understand what `pkg.f` should include. Could you please post its contents as well?

zhuazizhua, 2020-12-18:
@abdullahyildiz 
According to your code, you should use:"vlogan -full64 -nc -sverilog -assert svaext +define+WT_DCACHE $(filter %.sv,$(src)) +incdir+src/common_cells/include/+src/util/+incdir+$(VCS_HOME)/etc/uvm-1.2/macros"
"$(VCS_HOME) /etc/uVm-1.2/macros" requires that you modify the path that exists for the "uVM_glob_to_re" character of your VCS path, which you can find through the "grep" directive

zhuazizhua, 2020-12-18:
@abdullahyildiz
According to your code, you should use:"vlogan -full64 -nc -sverilog -assert svaext +define+WT_DCACHE $(filter %.sv,$(src)) +incdir+src/common_cells/include/+src/util/+$(VCS_HOME)/etc/uvm-1.2/macros"
"$(VCS_HOME) /etc/uVm-1.2/macros" requires that you modify the path that exists for the "uVM_glob_to_re" character of your VCS path, which you can find through the "grep" directive

zhuazizhua, 2020-12-18:
@abdullahyildiz
In my "pkg.f" file is "$(filter %. Sv,$(ariane_pkg))"

zhuazizhua, 2020-12-18:
In my "pkg.f" file is "$(filter %. Sv,$(ariane_pkg))",But I can't confirm your question appeared in "$(filter % sv, $(ariane_pkg)", "$(filter %.sv,$(util))" or"$(filter %.sv,$(src))".

abdullahyildiz, 2020-12-18:
Following lists where uvm_glob_to_re keyword exists in uvm-1.2 directory:

```
$ grep -r "uvm_glob_to_re" /eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_resource.svh:  //|    function string uvm_glob_to_re(string glob);
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_resource.svh:    scope = uvm_glob_to_re(s);
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_resource.svh:    re = uvm_glob_to_re(re);
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_globals.svh:  s = uvm_glob_to_re(expr);
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_config_db.svh:        if(uvm_re_match(uvm_glob_to_re(inst_name),w.inst_name) == 0)
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/dpi/uvm_regex.cc:// uvm_glob_to_re
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/dpi/uvm_regex.cc:const char * uvm_glob_to_re(const char *glob)
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/dpi/uvm_regex.svh:import "DPI-C" context function string uvm_glob_to_re(string glob);
/eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/dpi/uvm_regex.svh:function string uvm_glob_to_re(string glob);
```

Following are the commands that I run from Makefile:

```
vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2
vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 +define+WT_DCACHE $(filter %.sv,$(ariane_pkg)) +incdir+src/common_cells/include/+$(VCS_HOME)/etc/uvm-1.2/dpi
vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 +define+WT_DCACHE $(filter %.sv,$(util)) +incdir+src/common_cells/include/+src/util/+$(VCS_HOME)/etc/uvm-1.2/dpi
vhdlan -full64 $(filter %.vhd,$(uart_src))
vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 -assert svaext +define+WT_DCACHE $(filter %.sv,$(src)) +incdir+src/common_cells/include/+src/util/+$(VCS_HOME)/etc/uvm-1.2/dpi
vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 $(tbs)
vcs -full64 work.ariane_tb
```

Is this what you meant?

abdullahyildiz, 2020-12-18:
I fixed this problem by modifying the last command as `vcs -full64 -ntb_opts uvm-1.2 work.ariane_tb`. However, I get another error as:

```
Error-[DPI-DIFNF] DPI import function not found
/home/ayildiz/Desktop/cva6/tb/common/SimDTM.sv, 68
  The definition of DPI import function/task 'debug_tick' does not exist.
  Please check the stated DPI import function/task is defined, and its
  definition is either passed in a source file at compile-time, or provided in
  a shared library specified using the LRM Annex-J options at run-time.
```

zhuazizhua, 2020-12-18:
@abdullahyildiz 
"./simv -sv_lib work-dpi/ariane_dpi"I can only help you so far, because I'm trying to solve the next problem

zarubaf, 2020-12-18:
@abdullahyildiz The `debug_tick` function should be part of the `ariane_dpi.so`. It might be necessary to add the shared library to your call to `vcs`. At least that is necessary with Modelsim as we do not use the simulator to also compile the C sources.

abdullahyildiz, 2020-12-18:
@zhuazizhua, @zarubaf,

When I call VCS simulation as `./simv -sv_lib work-dpi/ariane_dpi`, I get a run-time error as:

htif: invalid option -- 's'
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Unknown argument (did you mean to enable +permissive parsing?)

Error-[DPI-UED] C++ Exception detected
  Import DPI routine invoked at file 
  '/home/ayildiz/Desktop/cva6/tb/common/SimDTM.sv'(line 68) has C++ exceptions
  not caught. C++ exceptions shall not propagate out of any imported 
  subroutine.
  Fix it in DPI-C code before running simulation.

zhuazizhua, 2020-12-18:
@abdullahyildiz @zarubaf 
sorry,I only retrieved "htilf" in "https://github.com/riscv/riscv-fesvr/blob/master/fesvr/htif.cc".Line 248.But I don't know how to solve it.

zarubaf, 2020-12-18:
> @zhuazizhua, @zarubaf,
>
> When I call VCS simulation as `./simv -sv_lib work-dpi/ariane_dpi`, I get a run-time error as:
>
> htif: invalid option -- 's'
> terminate called after throwing an instance of [...]

That actually means your simulation is working, that is the `htif` interface which is complaining about not understanding the `-sv_lib` parameter (because it is part of VCS and not `htif`).

Calling

```
./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off path/to/riscv/binary
```

should do the trick.

P.S.: As always, a pull request with the changes is very welcome! 😄

zarubaf, 2020-12-18:
> ```
> ./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off path/to/riscv/binary
> ```

The `+permissive` is ignored by VCS (because it is a plus-arg) and the `htif` interface will now that everything which comes afterward (until `+permissive-off`) is to be ignored.

abdullahyildiz, 2020-12-18:
It seems that I cannot pass the program path to HTIF with the following command:

```
$ ./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off rv64ui-p-addi

Warning-[RUNT-OPT-UNKNOWN] Unknown runtime option
  Unknown VCS runtime option 'rv64ui-p-addi' passed as part of command './simv
  +permissive -sv_lib work-dpi/ariane_dpi +permissive-off rv64ui-p-addi '.
  Please review valid vcs runtime options by executing ./<simulation 
  executable> -help

```

zarubaf, 2020-12-18:
My bad, you probably need to prepend two pluses (`+`) before the binary name:
```diff
- $ ./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off rv64ui-p-addi
+ $ ./simv +permissive -sv_lib work-dpi/aiane_dpi +permissive-off ++rv64ui-p-addi
```
Maybe see the call to `vsim` as an example:

https://github.com/openhwgroup/cva6/blob/6fd1a734e0a79a28b3965cb02a4df5ea399b39ab/Makefile#L344-L347

Please let me know whether this resolves your issue.

abdullahyildiz, 2020-12-18:
@zarubaf,

Thanks for the help.

I am still doing something wrong. Simulation starts but no progress is shown in terminal, also the simulation does not end:

```
./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off ++/eda/riscv/target/share/riscv-tests/isa/rv64ui-p-addi
Chronologic VCS simulator copyright 1991-2020
Contains Synopsys proprietary information.
Compiler version Q-2020.03-SP2_Full64; Runtime version Q-2020.03-SP2_Full64;  Dec 18 14:58 2020
UVM_INFO /eda/synopsys/vcs/Q-2020.03-SP2/etc/uvm-1.2/base/uvm_root.svh(412) @ 0: reporter [UVM/RELNOTES] 
----------------------------------------------------------------
UVM-1.2.Synopsys
(C) 2007-2014 Mentor Graphics Corporation
(C) 2007-2014 Cadence Design Systems, Inc.
(C) 2006-2014 Synopsys, Inc.
(C) 2011-2013 Cypress Semiconductor Corp.
(C) 2013-2014 NVIDIA Corporation
----------------------------------------------------------------

  ***********       IMPORTANT RELEASE NOTES         ************

  You are using a version of the UVM library that has been compiled
  with `UVM_NO_DEPRECATED undefined.
  See http://www.eda.org/svdb/view.php?id=3313 for more details.

  You are using a version of the UVM library that has been compiled
  with `UVM_OBJECT_DO_NOT_NEED_CONSTRUCTOR undefined.
  See http://www.eda.org/svdb/view.php?id=3770 for more details.

      (Specify +UVM_NO_RELNOTES to turn off this notice)
```

zarubaf, 2020-12-18:
That is interesting. The output looks like the simulation should be working. Just sanity checking:

1. Are you sure you waited long enough. The tests take some time (minutes) until completed.
2. Do you get a trace file? If you do have one, would you mind posting it? Maybe the end-of-computation mechanism is broken.

abdullahyildiz, 2020-12-18:
@zarubaf,

I waited a couple of minutes until I kill the simulation. I don't have trace file but there is a file named `uart`:

[image]

abdullahyildiz, 2020-12-22:
@zarubaf,

Just to make sure, I did the following modifications to the [Makefile](Makefile):

```
diff --git a/Makefile b/Makefile                                                                                                                                                                                                                                                
index e11c521..ac0e8eb 100644                                                                                                                                                                                                                                                   
--- a/Makefile                                                                                                                                                                                                                                                                  
+++ b/Makefile                                                                                                                                                                                                                                                                  
@@ -121,6 +121,7 @@ endif
 dpi_hdr := $(wildcard tb/dpi/*.h)                                                                                                                                                                                                                                              
 dpi_hdr := $(addprefix $(root-dir), $(dpi_hdr))                                                                                                                                                                                                                                
 CFLAGS := -I$(QUESTASIM_HOME)/include         \                                                                                                                                                                                                                                
+          -I$(VCS_HOME)/include               \                                                                                                                                                                                                                                
           -I$(RISCV)/include                  \                                                                                                                                                                                                                                
           -I$(SPIKE_ROOT)/include             \                                                                                                                                                                                                                                
           $(if $(DROMAJO), -I../tb/dromajo/src,) \                                                                                                                                                                                                                             
@@ -277,6 +278,15 @@ else
        questa-cmd += +jtag_rbb_enable=0                                                                                                                                                                                                                                        
 endif                                                                                                                                                                                                                                                                          
                                                                                                                                                                                                                                                                                
+vcs_build: $(dpi-library)/ariane_dpi.so                                                                                                                                                                                                                                        
+       vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2                                                                                                                                                                                                                          
+       vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 +define+WT_DCACHE $(filter %.sv,$(ariane_pkg)) +incdir+src/common_cells/include/+$(VCS_HOME)/etc/uvm-1.2/dpi                                                                                                             
+       vlogan -full64 -nc -sverilog -ntb_opts uvm-1.2 +define+WT_DCACHE $(filter %.sv,$(util)) +incdir+src/common_cells/include/+src/util/+$(VCS_HOME)/etc/uvm-1.2/dpi                                                                                                         
+       vhdlan -full64 $(filter %.vhd,$(uart_src))                                                                                                                                                                                                                              
[... cut ...]                                                             
 build: $(library) $(library)/.build-srcs $(library)/.build-tb $(dpi-library)/ariane_dpi.so                                                                                                                                                                                     
        # Optimize top level
```

I use Ubuntu 18.04.3 LTS and Synopsys VCS P-2019.06. I can reproduce the case if someone has successfully run the simulation in VCS.

yogurt1231, 2021-01-21:
@abdullahyildiz @zarubaf

yogurt1231, 2021-01-21:
@abdullahyildiz @zarubaf 

I have solved this problem, the reason for the simv simulation stuck is that the 'timescale' is not specified.

You should to add '`timescale 1ns/1ns' by pass over parameter or specify in source code (a new file or just add in ariane_tb.sv) when compiling with vcs, and then simv simulation should be proceed normally.

I have run a few cases in riscv-test, the simv can exit normally, and generate trace file.

If you need to dump waves, you need to call $fsdbDumpvars(). And you can control over it with $test$plusargs(). you should know how to do it if you are familiar with vcs.

Hope it helps, thank you for previous work.

abdullahyildiz, 2021-01-21:
@yogurt1231,

Thanks for your contribution. When I added _`timescale 1ns/1ns_ compiler directive to ariane_tb.sv, following command worked successfully and trace files were generated:

`./simv +permissive -sv_lib work-dpi/ariane_dpi +permissive-off ++/eda/riscv/cva6-toolchain/target/share/riscv-tests/isa/rv64ui-p-addi`

commit 95b1070a (2021-06-24): Add extended verification support (#685)

* Makefile, riscv_pkg.sv: Select C64A6 or CV32A6

according to variant variable

* add RVFI tracer and debug support

New files: rvfi_pkg.sv, rvfi_tracer.sv, ariane_rvfi.pkg.sv

- RVFI ports are added to ariane module
- rvfi_tracer.sv is a module added in ariane-testharness.sv
- RVFI_TRACE enables RVFI trace generation

* Move example_tb from cva6 to core-v-verif project

* Makefile: remove useless rule for vsim

* Makefile: add timescale definition when vsim is used

* Makefile: add vcs support (fix #570)

* rvfi_tracer.sv: fix compilation error raise [... cut]
commit 3ddf797e (2021-09-24): Re-organize CVA6 and APU (#725)

* Initial repository re-organization (#662)

Initial attempt to split core from APU.

Compile `corev_apu` (#667)

* Makefile verilates corev_apu
* Cleanup README
* Fix URL to repo
* Cleaned-up Makefile verilates corev_apu

Add extended verification support (#685)

* Makefile, riscv_pkg.sv: Select C64A6 or CV32A6

according to variant variable

* add RVFI tracer and debug support

New files: rvfi_pkg.sv, rvfi_tracer.sv, ariane_rvfi.pkg.sv

- RVFI ports are added to ariane module
- rvfi_tracer.sv is a module added in ariane-testharness.sv
- RVFI_TRACE enables RVF [... cut]
closed 2021-09-24 by zarubaf