openhwfoundation/cva6 #571: Make verilate exiting due to many errors in makefile

vasilistheod, 2020-12-11:
Hello everyone , i need to build the verilator model of cva6 (ariane).. i have followed all the instructions in the cva6 repository but when i do: make verilate so that i can build the verilator model it exits due to too many errors in the makefile.. i have searched a lot and i cant find the solution ..please someone help me , i would also appreciate if you suggested any other cores so that i can build easily the verilator mode
[image]
l

zarubaf, 2020-12-14:
On which Verilator version are you? I remember the default Ubuntu version to be _very_ old. I recommend at least some version `> 4.0`.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insight @zarubaf.

Hi @vasilistheod , thanks for your interest in CVA6. Personally, I use Verilator v4.100 for CVA6.  Since this issue has not been updated in more than two years, so I will assume it is resolved and will close this issue. If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup