openhwfoundation/cva6 #575: DC Synthesis Error with System Verilog

a7med7asan15, 2021-01-04:
[image]
When I try to analyze ariane_pkg.sv file it gives me those errors. 
clog2 is not synthesizable. 
The tool only has 2009, 2005 systemVerilog Standards.
Also I can't find dm package 
Thanks

zarubaf, 2021-01-04:
`clog2` should be synthesizable (given it evaluates to a constant expression). We normally do not have any problems with it (but using a recent DC version).

`dm` is here: https://github.com/pulp-platform/riscv-dbg/blob/6d768ac6371356b6e382c01ae7d860dd3e45248b/src/dm_pkg.sv

It is part of a submodule in the `src` directory. Did you checkout the submodules (i.e., `git submodule update --init --recursive`?

a7med7asan15, 2021-01-04:
Yes. I checked out them. 
What version do you usually use for systemVerilog standard in DC?

zarubaf, 2021-01-04:
I think it worked back to `2017.09` but we usually use the newest version, as SV support became much better.

a7med7asan15, 2021-01-04:
[image]
In this code here, Why clog2 does not see the parameter as constant?

zarubaf, 2021-01-04:
That code snippet should be guarded by the `pragma translate_off`, is the synthesizer complaining at that section?

a7med7asan15, 2021-01-04:
Yes. The synthesizer is complaining about it. I removed the function and the error is gone. Is it ok to do so?
But I have now other errors regarding fu_op op and riscv package
fu_op op, I think it doesn't recognize it and it gives me to put comma between them. 
regarding riscv package: I tried to analyze riscv_pkg.sv in include but it gave me same error that it needs riscv.pvk
[image]

zarubaf, 2021-01-04:
Yes, you need to have `riscv_pkg` before `ariane` package. My guess is that you have a partially valid database. I would always clean the design `remove_design -all` and also make sure that the work folder (or whatever folder holds the `*.pvk`) is deleted.

zarubaf, 2021-01-04:
Also removing the function will likely not work, just remove the body (or better just set the `VERILATOR` define). It is very strange though, that your Synthesiser doesn't obey the `pragma translate_off`. Haven't seen such a case and I suspect that there is something fishy with your setup.

a7med7asan15, 2021-01-04:
Same error. 
Also didn't differ when I uncommented `pragma translate_off` 
Here is the error when analyzing riscv_pkg
[image]

zarubaf, 2021-01-04:
It is basically the same problem, it doesn't obey the `pragma` and fails to analyze the package (see the last errors):

https://github.com/openhwgroup/cva6/blob/0bbc3cb5cf43cd3bf77d3872e46dd0d6ededf28f/include/riscv_pkg.sv#L650-L674

That is why the `riscv_pk`  does not analyze successfully. I suggest you dig into the user manual and see why or how the synthesizer should obey the pragma. As far as I remember there wasn't anything special to be set (might depend on the Synthesiser version). You need to fix this issue, there a handful of non-synthesizable construct guarded in this way in most of our codebase.

a7med7asan15, 2021-01-04:
I found the solution for it. Instead of pragma, It should use the synthesizer's program name. 
DC Compiler: // synopsys translate_on
RC: // cadence translate_on/off
Thanks a lot for your help

a7med7asan15, 2021-01-04:
Still gives me error with riscv package. Does it mean another riscv package rather than riscv_pkg in include? 
Same issue when running ariane_pkg
[image]

zarubaf, 2021-01-05:
No, that is the only one.

a7med7asan15, 2021-01-05:
I think the problem is here: 
Why use riscv:: while it's already in riscv package? shouldn't this be removed?
When I removed riscv:: the synthesizer worked ok
[image]

MikeOpenHWGroup, 2023-02-17:
Thanks for your insight @zarubaf.

Hi @a7med7asan15, thanks for your interest in CVA6. This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue. If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup

Zsmailing, 2025-05-20:
May I ask if it has been solved? I encountered the same problem

[image]

Zsmailing, 2025-05-20:
Thanks，the problem has been found.