openhwfoundation/cva6 #579: benchmarks baremetal execution failed

RaphaelKlink, 2021-01-13:
Hey,

i tried to run the Dhrystone and Coremark baremetal on the CVA6 core on a FPGA. I get stuck in the init system call. How can I run these benchmarks in baremetal?

Thank you for your help.

Greetings
Raphael Klink

niwis, 2021-01-18:
Hello Raphael, the riscv-tests use a front-end server to delegate system calls from CVA6 to the host. This functionality is only available in simulation and not on FPGA. Unfortunately, a baremetal execution environment is not provided at the moment - you would need to create your own or compile the applications standalone.
Best, Nils

MikeOpenHWGroup, 2023-02-17:
Thanks for your insights @niwis.

Hi @RaphaelKlin,  thanks for your interest in CVA6.  This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup