openhwfoundation/cva6 #588: Atomic instructions stalling at atomic instructions

Rahul-Kande, 2021-01-18:
Hello, 

I am simulating atomic instructions on baremetal ariane. But, the processor is getting stuck at the atomic instruction and does not simulate anything else once it starts executing atomic instructions. 

I looked into the issue and figured out a few things: 
1) This is happening with all atomic instructions other than atomic load and store. 
2) This only happens when the value in the rs1 register is zero. 
3) I observed that the atomic instruction reaches the commit stage module but does not get committed since the amo_resp_i.ack signal doesnt go high. 

I am attaching the C program for amoxor and corresponding log files. Please help me fix this issue
[atomic_instruction_issue.zip](https://github.com/openhwgroup/cva6/files/5832085/atomic_instruction_issue.zip)

zarubaf, 2021-01-19:
You aber trying to load from unmapped memory at address 0, hence, the processor is stalling. `rs1` contains the address of the atomic. Also note that atomics are only supported on main memory.

closed 2021-01-19 by zarubaf

Rahul-Kande, 2021-01-19:
But shouldn't the processor raise an exception or something? 
Currently it is stuck in that instruction forever which should not be the case for any instruction no?