openhwfoundation/cva6 #589: error : work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf

ekfkawnlvn, 2021-01-19:
when I follow this example they make some error message

```
~/WORK/Ariane/temp/ariane$ work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf

**> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 42875**
```

someone told me wait more time, but i'm waiting more than 2hour...

i'm using ariane-dev branch

zarubaf, 2021-01-19:
I‘d suggest you use the master branch, the `dev` branch is considered unstable. The command is correct, I think it takes in the order of 10ish minutes. Let me know if it works for you on the current master.

ekfkawnlvn, 2021-01-20:
@zarubaf 
Can I use any verilator version? because yesterday i was using v4.107 but master-branch isn't running

zarubaf, 2021-01-20:
https://github.com/openhwgroup/cva6#tool-requirements

Yes, I know. But I believe that there is something wrong with Verilator in the latest version(s). Also other projects don’t run.

tl;dr stick to v4.100

ekfkawnlvn, 2021-01-20:
@zarubaf 
Thank you for your prompt reply.
I've confirmed that it's going well with version 4.034.

closed 2021-01-20 by ekfkawnlvn

NazerkeT, 2021-06-03:
> @zarubaf
> Thank you for your prompt reply.
> I've confirmed that it's going well with version 4.034.

Dear @ekfkawnlvn could you please share your GCC version and OS? I am having a similar issue, so wanted to know more. 

Thanks, Best wishes,