openhwfoundation/cva6 #590: synthesis fails on Vivado 2020.2

monniaux, 2021-01-20:
eaeef7d

RTL Elaboration failed
386 Infos, 52 Warnings, 0 Critical Warnings and 82 Errors encountered.
synth_design failed
[fpga.log](https://github.com/openhwgroup/cva6/files/5844507/fpga.log)

leemango1998, 2021-01-21:
I also meet this error. But when I open the ariane.xpr project with vivado GUI and run synthesis,  it succeed. And I have finished the implementation.

monniaux, 2021-01-21:
I tried running synthesis in Vivado GUI using ariane.xpr and I got.

[Common 17-55] 'set_property' expects at least one object.
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.

monniaux, 2021-01-21:
`make fpga` succeeds with Vivado 2019.2.1

zarubaf, 2021-02-01:
Hm, I think you somehow didn't synthesize the IPs (or with an older version):
```
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Synthesis target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_mig_7_ddr3/ip/xlnx_mig_7_ddr3.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_clk_gen/ip/xlnx_clk_gen.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_clock_converter/ip/xlnx_axi_clock_converter.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_gpio/ip/xlnx_axi_gpio.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_dwidth_converter/ip/xlnx_axi_dwidth_converter.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_quad_spi/ip/xlnx_axi_quad_spi.xci

WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Implementation target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_mig_7_ddr3/ip/xlnx_mig_7_ddr3.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_clk_gen/ip/xlnx_clk_gen.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_clock_converter/ip/xlnx_axi_clock_converter.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_gpio/ip/xlnx_axi_gpio.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_dwidth_converter/ip/xlnx_axi_dwidth_converter.xci
/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/xilinx/xlnx_axi_quad_spi/ip/xlnx_axi_quad_spi.xci
```

The warning seems strange and related to the first error:
```
ERROR: [Synth 8-439] module 'xlnx_axi_dwidth_converter' not found [/home/monniaux/progs/Risc-V/CVA6/cva6/fpga/src/ariane_peripherals_xilinx.sv:351]
```

Can you try removing the `fpga/work-fpga/` folder (i.e., `rm -r fpga/work-fpga/`)?

zarubaf, 2021-02-01:
Just confirming on my end that the synthesis runs fine on Vivado 2020.1

monniaux, 2021-02-01:
I did a git clean and `make fpga` failed on Vivado 2020.2.

zarubaf, 2021-02-01:
Okay, I can also confirm the error with 2020.2. I think I've found the problem. Just testing this on different Vivado versions now.

zarubaf, 2021-02-01:
Potential fix in #604, if you want to try it out early :-)

hafssa-hassy, 2021-11-23:
> I also meet this error. But when I open the ariane.xpr project with vivado GUI and run synthesis, it succeed. And I have finished the implementation.

I am sorry but where can I find the ariane.xpr project?

leemango1998, 2021-11-23:
> 

The path of the CVA6 project has changed recently, and I guess the current path of ariane.xpr should be cva6/corev_apu/fpga/  after you make fpga.

hafssa-hassy, 2021-11-30:
> `make fpga` succeeds with Vivado 2019.2.1
I tried 'make fpga' with the two versions 20219.1 and 2020.2 but it gives the same error.
vivado -mode batch -source tcl/run.tcl

****** Vivado v2020.2 (64-bit)
  **** SW Build 3064766 on Wed Nov 18 09:12:47 MST 2020
  **** IP Build 3064653 on Wed Nov 18 14:17:31 MST 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
no such variable
    (read trace on "::env(XILINX_PART)")
    invoked from within
"set partNumber $::env(XILINX_PART)"
    (file "tcl/run.tcl" line 1)
INFO: [Common 17-206] Exiting Vivado at Wed Dec  1 00:21:51 2021...
make[1]: *** [../common.mk:2: all] Error 1
make[1]: Leaving directory '/tp/xph3seo/xph3seo508/Documents/cva6/corev_apu/fpga/xilinx/xlnx_axi_clock_converter'
make: *** [Makefile:34: work-fpga/xlnx_axi_clock_converter.xci] Error 2

*******************************************************************************************************

vivado -mode batch -source tcl/run.tcl

****** Vivado v2019.1 (64-bit)
  **** SW Build 2552052 on Fri May 24 14:47:09 MDT 2019
  **** IP Build 2548770 on Fri May 24 18:01:18 MDT 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source tcl/run.tcl
no such variable
    (read trace on "::env(XILINX_PART)")
    invoked from within
"set partNumber $::env(XILINX_PART)"
    (file "tcl/run.tcl" line 1)
INFO: [Common 17-206] Exiting Vivado at Wed Dec  1 00:12:16 2021...
make[1]: *** [../common.mk:2: all] Error 1
make[1]: Leaving directory '/tp/xph3seo/xph3seo508/Documents/cva6/corev_apu/fpga/xilinx/xlnx_axi_clock_converter'
make: *** [Makefile:34: work-fpga/xlnx_axi_clock_converter.xci] Error 2

MikeOpenHWGroup, 2023-02-17:
Hi @monniaux, @hafssa-hassy and @leemango1998,  thanks for your interest in CVA6.  This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup