openhwfoundation/cva6 #591: Out of date instructions in the README

MikeOpenHWGroup, 2021-01-20:
I believe the instruction for obtaining the required toolchain in the [Getting Statted](https://github.com/MikeOpenHWGroup/cva6#getting-started) of the README are out of date.  According to [riscv-gnu-toolchain #174](https://github.com/riscv/riscv-gnu-toolchain/issues/174#issuecomment-508962471):
> riscv-tools no longer includes a toolchain. You must build a riscv toolchain before you can build riscv-tools.

It would be a good idea to update the README accordingly.

zarubaf, 2021-01-29:
Yes, I agree. Unfortunately one still needs the [`fesvr`](https://github.com/riscv/riscv-isa-sim/) which is part of the toolchains repo, hence the instructions in the toolchain repo are still somehow. Any input on how we can make things simpler is appreciated.

MikeOpenHWGroup, 2021-01-29:
> Unfortunately one still needs the fesvr which is part of the toolchains repo

This is a sub-topic of a much larger conversation about how the CVA6 RTL model, its simulation testbench, FPGA emulation and toolchain are all interdependent.   @zarubaf, I will be opening a discussion about that with you and @JeanRochCoulon later today.

For this issue, I recommend expanding the Getting Started section into two paragraphs, one with instructions to obtain and install the toolchain, and another with instructions to obtain and install the riscv-isa-sim.  Also, maybe get rid of the "fesvr" name... nobody says "RISC-V Front-End Server" anymore.  :stuck_out_tongue_winking_eye:

zarubaf, 2021-01-29:
Fyi: These are two different things: `fesvr` is the host part for `riscv-isa-sim`, we do not need `spike` but we need the hosting environment (also for simulation of the debug module). Unfortunately, it was decided to merge those two things into `riscv-isa-sim` repo so that is why we need it. Originally, it has been a separate repository.

MikeOpenHWGroup, 2021-01-29:
Thanks for the info.  I think you've just proved my point: the README needs an update.

zarubaf, 2021-01-29:
Happy to accept a PR.

MikeOpenHWGroup, 2021-01-29:
> Happy to accept a PR.

Not happening my friend.  This is a bug report, and I am not qualified to fix it.   It will be the same with the RTL: bugs will be reported via GitHub issues and a member of your design team will need to implement a fix and generate the PR to resolve it.

JeanRochCoulon, 2021-01-29:
Hello @zarubaf @MikeOpenHWGroup @ASintzoff,
There is an alternative way to obtain the required toolchain. By following
https://github.com/openhwgroup/core-v-verif/blob/master/cva6/README.md, you will be redirect to https://github.com/ThalesGroup/cva6-tools, then you have to execute commands listed in https://github.com/ThalesGroup/cva6-tools/blob/master/toolchain/README.md
Tell me if it is better(or not better for sure)
Regards

zarubaf, 2021-01-29:
Thanks, @JeanRochCoulon for stepping in. What I couldn't find out is if it also installs the `fesvr` in a `$RISCV` directory. That would be needed with the simulation environment shipped with this repository.

There is also a pre-built toolchain available from SiFive. This is what the CI is using and what I am using currently tbh.

JeanRochCoulon, 2021-01-29:
In cva6-tools, gcc is built (executing the README pointed above). fesvr is installed when setting up the core-v-verif env, it is built together with SPIKE, a env variable is dedicated to point on the binaries.
I have preferred to build our tools to be able to add the CVA6 architecture (-march=CVA6, nice is not it?), to build for the RV32 and RV64 variant (usefle to compile to RV32A6 and RV64A6) and to supply the multilibs. In the near futur, we will use the same infrastructure to support also LLVM.

MikeOpenHWGroup, 2021-01-29:
I like this @JeanRochCoulon.   In any case, its important that we standardize on a toolchain and publish the instructions for getting it.  Once we get a lot of Contributors it will become messy if this is not in place.

JeanRochCoulon, 2021-01-29:
@MikeOpenHWGroup and @zaruba, you are the best guys to validate (or feedback drawbacks) the compile infrastructure. Tell me. CVA6-tools could be transferred if needed to openHW. As this project is a "TOP" project which calls to GCC, binutils, newlib,... so it consists in a script which calls the build of all the subprojects, that's why maintenance is limited.

MikeOpenHWGroup, 2021-01-29:
OK.  I'm actually in a pretty good spot to validate this now.  Let's move this discussion to MatterMost and post conclusions here later.

MikeOpenHWGroup, 2023-02-17:
Whew!   Mon bad.  This issue was fixed almost two years ago and I have not closed it.

closed 2023-02-17 by MikeOpenHWGroup