openhwfoundation/cva6 #596: Seeing error in ariane_tb with questasim

jatanpandya83, 2021-01-28:
# Optimize top level
vopt +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive +define+WT_DCACHE -work work  ariane_tb -o ariane_tb_optimized +acc -check_synthesis
** Error (suppressible): tb/ariane_tb.sv(149): (vopt-7063) Failed to find 'genblk1' in hierarchical name 'dut.i_sram.genblk1[0].genblk1.i_ram.Mem_DP'.
        Region: ariane_tb
Optimization failed

I couldn't find genblk1 in src/util/sram.sv. Where is genblk1? Thanks!

zarubaf, 2021-01-28:
Oops, that is indeed a bad mistake. Sorry. Apparently I forgot to update the questasim testbench. There is a fix for it in #597, thanks for noticing it.

jatanpandya83, 2021-01-28:
Solved. thanks!

closed 2021-01-28 by jatanpandya83