openhwfoundation/cva6 #599: SRAM question, potential [BUG]

sotot0, 2021-01-29:
Due to observation by me and a colleague (Iasonas Mastorakis), the input be_i of SRAM module is referred to **bit enable**. Although, the input coming from TAG_CMP module, be_ram, is referred to **byte enable**. Because of that, simulation of the behavioural SRAM leads to 'Z values on the most of MS bits. In this way, it is impossible to write the whole cache line in SRAM. 
Please confirm that the potentiality of this bug.

Greetings

niwis, 2021-03-01:
Hello @sotot0,

I believe what you are describing is explained here: https://github.com/openhwgroup/cva6/blob/7760227f887c59a2ae1b89e70fb20582a2d5c5b3/src/cache_subsystem/std_nbdcache.sv#L201-L203
So each byte in the `valid_dirty_sram` represents one cache line: bit `0` is the dirty bit, bit `1` is the valid bit, and bits `2` to `7` of each byte are unused. The byte alignment is done s.t. the valid/dirty bits of each cache line can be updated individually using the SRAM's byte enable signal. Does this answer your question?

Best,
Nils

MikeOpenHWGroup, 2023-02-17:
Hi @niwis, thanks for your insights.

Hi @sotot0,  thanks for your interest in CVA6.  This issue has not been updated in almost two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup