openhwfoundation/cva6 #601: user program not loading correctly using questa sim

jatanpandya83, 2021-01-31:
Hi,

I am trying to run this test "rv64ui-p-sraw". The test finishes successfully, but after comparing the .dump with the trace_log, I realized that the simulation did not even read the user program from the DRAM location: 80000000. It seems that the user program was not loaded at all in the DRAM.

Here is the commandline that I use:
make sim elf-bin=tmp/riscv-tests/build/isa/rv64ui-p-sraw batch-mode=1 

Here is a snippet of the trace log output:
    5490ns      267 M 0000000000010000 0 00100413 li             s0, 1                 s0  :0000000000000001
    5590ns      272 M 0000000000010004 0 01f41413 slli           s0, s0, 0x1f          s0  :0000000080000000 s0  :0000000000000001
    5690ns      277 M 0000000000010008 0 f1402573 csrr           a0, mhartid           a0  :0000000000000000
    5790ns      282 M 000000000001000c 0 00000597 auipc          a1, 0x0               a1  :000000000001000c
    5890ns      287 M 0000000000010010 0 07458593 addi           a1, a1, 116           a1  :0000000000010080 a1  :000000000001000c
    5990ns      292 M 0000000000010014 1 00040067 jalr           x0, s0, 0             s0  :0000000080000000
**_**_Exception @      6190, PC: 0000000080000000, Cause: Illegal Instruction_**,_** 
				tval: 0000000000000000
    6390ns      312 M 0000000000010040 0 f1402573 csrr           a0, mhartid           a0  :0000000000000000
    6490ns      317 M 0000000000010044 0 00000597 auipc          a1, 0x0               a1  :0000000000010044
    6590ns      322 M 0000000000010048 0 03c58593 addi           a1, a1, 60            a1  :0000000000010080 a1  :0000000000010044

It shows that while executing the bootrom, when PC=80000000, it  results in an illegal instruction(as above), instead of jumping to the reset vector instruction.

Here is a snippet of the .dump, which I expected:
Disassembly of section .text.init:

0000000080000000 <_start>:
    80000000:	04c0006f          	j	8000004c <reset_vector>

0000000080000004 <trap_vector>:
    80000004:	34202f73          	csrr	t5,mcause
    80000008:	00800f93          	li	t6,8
    8000000c:	03ff0a63          	beq	t5,t6,80000040 <write_tohost>
    80000010:	00900f93          	li	t6,9
    80000014:	03ff0663          	beq	t5,t6,80000040 <write_tohost>
    80000018:	00b00f93          	li	t6,11
    8000001c:	03ff0263          	beq	t5,t6,80000040 <write_tohost>
    80000020:	80000f17          	auipc	t5,0x80000
    80000024:	fe0f0f13          	addi	t5,t5,-32 # 0 <_start-0x80000000>
    80000028:	000f0463          	beqz	t5,80000030 <trap_vector+0x2c>
    8000002c:	000f0067          	jr	t5
    80000030:	34202f73          	csrr	t5,mcause
    80000034:	000f5463          	bgez	t5,8000003c <handle_exception>

Any help is appreciated! 

Thanks,
Jatan P.

zarubaf, 2021-01-31:
I am pretty sure the program gets executed correctly. The debug module is being used to load the user program to main memory. Once that is done the control flow is redirected to the entry address of the ELF. The initial exception is just an artifact of the fact that there is no program loaded after reset. 

If you search the trace for further occurrences of the address of the start Symbol you will find that it it executes correctly.

One further note: Even the execution of the program will be periodically be interrupted by the debug module to check for end of computation.

jatanpandya83, 2021-01-31:
I did search through the trace log, but couldn't find the instruction at the reset vector getting executed. I have attached the file here. 
Thanks @zarubaf  for your prompt response. I really appreciate all the help!
[tracelog.txt](https://github.com/openhwgroup/cva6/files/5900506/tracelog.txt)

jatanpandya83, 2021-02-02:
After inspecting carefully, I realized that it is working. Thank you!

closed 2021-02-02 by jatanpandya83