openhwfoundation/cva6 #603: Describe CVA6 for write-through and write-back cache versions

MikeOpenHWGroup, 2021-02-01:
In #602, there was a conversation like this:

> Is the list of files in the manifest complete?
>> Looks okay for the write-through cache version. Of course, the write-back cache needs different files.

AFAICT, the ability to compile CVA6 for either  write-through and write-back cache versions is not described in the User Manual.  The User Manual should be updated to explain:
1. How to compile the CVA6 for  write-through and write-back cache versions.
2. Theory of operation for the write-through cache controller.
3. Theory of operation for the write-back cache controller.

j-silv, 2022-02-26:
> 1. How to compile the CVA6 for  write-through and write-back cache versions.

For an FPGA emulation (say for example, using the [Genesys 2 board](https://reference.digilentinc.com/reference/programmable-logic/genesys-2/reference-manual)) can we enable the write-back cache by simply commenting out the `` `define WT_DCACHE`` line in [the genesysii.svh file?](https://github.com/openhwgroup/cva6/blob/3ddf797e95923fd11113c8e443046105dfbf8843/corev_apu/fpga/src/genesysii.svh)

It seems as though the `WB_DCACHE` define is not mentioned anywhere else in the source code, so I'm assuming we don't need to uncomment that line.

suppamax, 2022-11-08:
@j-silv yes, by default the write-back cache is selected; you have compile the design with `WT_DCACHE` defined if you want to use a write-through cache

suppamax, 2022-11-08:
We've just [published](https://planv.tech/2022/11/08/cva6s-instruction-and-writeback-data-cache/) our notes about the WB cache.
And here's an interesting [paper](https://arxiv.org/pdf/2202.03749.pdf) about the WT cache.

Hope it helps

MikeOpenHWGroup, 2023-02-17:
Hi @JeanRochCoulon, this is an old issue with new information.  Let's get this into the User Manual.

JeanRochCoulon, 2023-03-14:
I agree. The cache type should be addded in the CVA6 parameter chapter of user manual. I take the point !

JeanRochCoulon, 2023-03-21:
Hi, I have just merged the #1127 to replace the define by a localparam into the config pkg. On my side, the issue is solved. Mike, up to you to take the decidsion to close the issue.

MikeOpenHWGroup, 2023-03-23:
Thanks everyone!  Closing.

closed 2023-03-23 by MikeOpenHWGroup