openhwfoundation/cva6 #605: object dump for hello.elf does not contain the instructions needed at reset_vector

jatanpandya83, 2021-02-03:
The object dump generated from the hello.elf does not contain the instructions at reset vector, like the csr writes to mtvec, hart_id, etc. It seems that the program directly jumps to the user program. To compile hello.elf, I ran the following:

>riscv64-unknown-elf-gcc hello.c -o hello.elf

Do I need to specify the crt.S while compiling? Or is there something else that I need to do, to fix this? Please let me know!

I have the objdump attached!
[hello.txt](https://github.com/openhwgroup/cva6/files/5915269/hello.txt)

Thanks,
Jatan

jatanpandya83, 2021-02-08:
Any help is appreciated! Thanks!

zarubaf, 2021-02-09:
If you do not specify a crt0 the dafult libgloss one is selected. I am not sure I get your question. Which crt0 would you like to see used?

jatanpandya83, 2021-02-09:
As I mentioned, the object dump from hello.c does not contain the initial instructions at reset vector, like csr write to mtvec, hart_id, etc. It jumps directly to the user program, which is a problem. Is there anything that I need to specify while compiling my C program? 

Thanks,
Jatan

Yi1101, 2021-02-10:
> riscv64-unknown-elf-gcc -I./../env -I./common -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o xxx.riscv xxx.c ./common/syscalls.c ./common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T ./common/test.ld

I used the aboved compiled command, it can set the entry point to 800000, you are also find this inside the Makefile.

jatanpandya83, 2021-02-10:
Thanks @Yi1101 and @zarubaf . I believe it is working now for me, with the test finishing successfully! Appreciate all the help.

closed 2021-02-10 by jatanpandya83