openhwfoundation/cva6 #608: Linux boot takes few minutes on Genesys2 BOARD

melmajih, 2021-02-05:
Hello, 

Linux boot time on Genesys Board is critical (about 3 min). I noticed that the core frequency is 50 MHz but I am wondering if it is the cause or not? Could you tell if you have already measured linux boot time? 
Thnaks in advance 

Yours 
Mustapha

zarubaf, 2021-02-05:
The boot is actually taking a couple of seconds (around 30 in my experience). The slow part is loading the first-stage bootloader plus Linux kernel from SPI-Mode SD Card to main memory.

The Ariane+Openpiton [image](http://www.princeton.edu/~cloud/openpiton/bitfiles/openpiton_ariane_genesys2_1core_r13.bit) has an SD controller that supports higher-speeds. There you can see that the loading is significantly faster.

We could probably add the SD controller here as well but usually everything except the SPI mode needs licensing, at least that was my understanding. Not sure how they cope with that in OpenPiton.

melmajih, 2021-02-05:
Thanks a lot for your quick reply. Exactly the boot takes few seconds (could bu justified by a 50 MHz CPU clock) but if we include the step of loading the first-stage bootloader, it takes about 3 min. I have tested both ariane+openpiton and ariane (release 4.2) and it is the same thing. The clock of SPI in the microsd is 12.5 MHz. So as you said i need a SD controller which is more adapted for high speed. 
We want to use Ariane core in a project but boot time is critical for our use case, so if you have any solutions (open source) it will be great for us. 
Yours 
Mustapha

closed 2021-03-02 by melmajih

Steinegger, 2021-06-16:
> We could probably add the SD controller here as well but usually everything except the SPI mode needs licensing, at least that was my understanding. Not sure how they cope with that in OpenPiton.

This [vivado-risc-v repo](https://github.com/eugene-tarassov/vivado-risc-v/tree/master/sdc) seems to be using a faster and open source Verilog SD-card implementation. According to the readme of the repo it offers`SD HS (High Speed) specs, 25MB/s read/write speed`. Might be interesting to check it out and replace the SPI implementation in this repo if possible.

zarubaf, 2021-06-16:
That would be terrific indeed. Unfortunately, the mentioned core uses LGPL which isn't compatible with the liberal license we use here so we would need to re-write or the original author would need to re-license.

zarubaf, 2021-06-16:
Ah, and we'd need to make sure that we have the right Linux drivers :-)

Steinegger, 2021-06-16:
I'm not a lawyer but couldn't LGPL cores be used if unmodified?  Regardless of that it appears that the maintainer is also the author of that core, so maybe opening an issue there, asking him nicely to provide it standalone with a compatible license could work.

It boots directly to linux from an SD-card so there's to be a driver _somewhere_.

zarubaf, 2021-06-17:
We kicked off a similar discussion and seems that those are not compatible. Asking nicely seems like a very good approach.