openhwfoundation/cva6 #611: Can not compile CVA6 for Verilator, ReadMe still errant

gabaxter, 2021-02-11:
Hello, 

The readme file does NOT work, and I am not precisely sure what is failing.  Seems a missing include. Whereas the tool chains themselves are poorly documented, I followed the method proffered by @JeanRochCoulon in _Out of date instructions in the README #591_, which eventually takes you to following: https://github.com/ThalesGroup/cva6-tools/tree/master/toolchain.  Once that is done, Which BTW has issues in terms of accurately describing how to use and assumes things it should not, then you end up with tools installed at `$INSTALL_DIR`, which you then have to reset your `$RISCV` to point to, and then things like `$RISCV/bin/riscv-none-elf-gcc` etc are available.  Great, first significantly major hurdle over.

**Next, we do the install** 
`git clone https://github.com/openhwgroup/cva6.git`
`git submodule update --init --recursive`

**Then we do a make, and it fails...**
`make verilate`

**After a bit, we get this error - which I've seen consistently no matter how /what tools are installed:**
`make[1]: Entering directory '/home/glenn/dev/cva6/work-ver'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/home/glenn/dev/riscv-tools/toolchain/include -I/home/glenn/dev/riscv-tools/toolchain/include  -std=c++11 -I../tb/dpi   -c -o ariane_tb.o ../tb/ariane_tb.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/home/glenn/dev/riscv-tools/toolchain/include -I/home/glenn/dev/riscv-tools/toolchain/include  -std=c++11 -I../tb/dpi   -c -o SimDTM.o ../tb/dpi/SimDTM.cc
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Variane_testharness.cpp Variane_testharness___024unit.cpp Variane_testharness_uart_bus.cpp Variane_testharness_REG_BUS__A20_D20.cpp Variane_testharness_AXI_BUS__A40_AB40_AC5_AD1.cpp Variane_testharness__Dpi.cpp > Variane_testharness__ALLfast.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Variane_testharness__Syms.cpp > Variane_testharness__ALLslow.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/home/glenn/dev/riscv-tools/toolchain/include -I/home/glenn/dev/riscv-tools/toolchain/include  -std=c++11 -I../tb/dpi   -c -o Variane_testharness__ALLfast.o Variane_testharness__ALLfast.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I/include -I/home/glenn/dev/riscv-tools/toolchain/include -I/home/glenn/dev/riscv-tools/toolchain/include  -std=c++11 -I../tb/dpi   -c -o Variane_testharness__ALLslow.o Variane_testharness__ALLslow.cpp
../tb/dpi/SimDTM.cc:4:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
../tb/ariane_tb.cpp:34:10: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
          ^~~~~~~~~~~~~
compilation terminated.
Variane_testharness.mk:66: recipe for target 'ariane_tb.o' failed
make[1]: *** [ariane_tb.o] Error 1
make[1]: *** Waiting for unfinished jobs....
Variane_testharness.mk:68: recipe for target 'SimDTM.o' failed
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory '/home/glenn/dev/cva6/work-ver'
Makefile:478: recipe for target 'verilate' failed
make: *** [verilate] Error 2
`

It seems the `#include <fesvr/dtm.h>` is not able to be found.  At this point I am completely stuck, and there seems no google out of it ;)

Please advise how to resolve.

ASintzoff, 2021-02-11:
The file `fesvr/dtm.h` is part of `spike` installation. I don't know exactly how to manage it with the configuration you have.

To resolve your issue, I would suggest you to follow the README.md at https://github.com/openhwgroup/core-v-verif/blob/master/cva6/README.md
This explains how to execute tests on different CVA6 configurations using riscv-tests and riscv-compliance test suites.
If you don't want to run all the tests, you can only execute `source cva6/smoke-tests.sh` instead of `source cva6/dv-riscv-compliance.sh` or `source cva6/dv-riscv-tests.sh`

BTW, could you open an issue on https://github.com/ThalesGroup/cva6-tools to describe the problems you identified?

Thank you

gabaxter, 2021-02-11:
Hi André,

Thank you for the help.  There is a larger issue I pointed out in another issue which is the docs are simply out of date and wrong.  Since the tool chains have been removed from this repo, the repo no longer functions.  Good docs can fix this.  Someone should take a blank Ubuntu VM and get the directions required to make that function.  Then anyone can know they have a set of tools that work.  I have spike installed in my machine, but because different repos use different tools, trying to evaluate any given RISCV solution becomes nearly impossible.  I've used four different repos so far. 0% of them work 'out of the box' following readme.md.  0.   This is not good if we want the community to adopt.  

WRT specific issue here, I will try to install spike into this repo install and see if I can resolve.  But what is needed is a correct set of directions with a specific tool chain that fully supports this repo.  These are issues that developers will not see because their environment exists, but which anyone new to the repo will see every time.  In which case, they stop use the repo and go to another.  

I will try to open a issue on the other link as well.  I did not document what I did to get around them, but I may still have my ssh history on the server I was using.

Thank you for your help in identifying this issue.  Hopefully spike install and now the repo fully works.

dpetrisko, 2021-04-17:
For anyone else struggling with this, this patch fixes the fesvr include issue:

```
diff --git a/Makefile b/Makefile
index 4572870..f20f618 100644
--- a/Makefile
+++ b/Makefile
@@ -561,6 +561,7 @@ verilate_command := $(verilator)
                     $(if $(DEBUG),--trace --trace-structs,)                                                      \
                     -LDFLAGS "-L$(RISCV)/lib -L$(SPIKE_ROOT)/lib -Wl,-rpath,$(RISCV)/lib -Wl,-rpath,$(SPIKE_ROOT)/lib -lfesvr$(if $(PROFILE), -g -pg,) $(if $(DROMAJO), -L../tb/dromajo/src -ldromajo_cosim,) -lpthread" \
                     -CFLAGS "$(CFLAGS)$(if $(PROFILE), -g -pg,) $(if $(DROMAJO), -DDROMAJO=1,) -DVL_DEBUG"       \
+                    -CFLAGS "-I $(RISCV)/riscv-isa-sim/"                                                         \
                     -Wall --cc  --vpi                                                                            \
                     $(list_incdir) --top-module ariane_testharness                                               \
                     --Mdir $(ver-library) -O3                                                                    \
```

MikeOpenHWGroup, 2023-02-17:
Hi @ASintzoff, @dpetrisko, thanks for your insights.

@gabaxter, thanks for your interest in CVA6.  This issue has not been updated in almost two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another

closed 2023-02-17 by MikeOpenHWGroup