openhwfoundation/cva6 #615: Error: invalid ELF file, only 32bits files are supported

jatanpandya83, 2021-02-18:
When I try to  load 64 bit elf image through OpenOCD & bitbang, for Ariane tb in simulation, I see the following error: "Error: invalid ELF file, only 32bits files are supported".  Since Ariane core is 64 bit target, how do I resolve this issue? Is there a known fix? 

Thanks,
Jatan

zarubaf, 2021-02-18:
I’ve never seen this problem. May it be that’s you compiled GDB with only 32 bit support?

jatanpandya83, 2021-02-18:
I am using Telnet to load the image. For the image, I am using precompiled elf files from riscv-tests. Below is the command that I use after establishing a telnet connection with the localhost:
>load_image rv64ui-p-sraw

jatanpandya83, 2021-02-18:
OpenOCD only works with 32bit targets at the moment! This is a fundamental limitation:
http://openocd.org/doc/doxygen/html/targetnotarm.html#:~:text=64%20bit,only%20supports%2032%20bit%20targets.

Closing the issue now. Thanks!

closed 2021-02-18 by jatanpandya83