openhwfoundation/cva6 #633: After boot completed of the cva6 core,the GDB Read and write questions?

gzhan88, 2021-03-17:
########################################
##
##First:The serial port prints information
##
########################################
Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD... 
SD command cmd0         response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 00
sd initialized!
gpt partition table header:
        signature:      5452415020494645
        revision:       00010000
        size:           0000005C
        crc_header:     672B50E8
        reserved:       00000000
        current lba:    0000000000000001
        backup lda:     0000000003AFBFFF
        partition entries lba:          0000000000000002
        number partition entries:       00000080
        size partition entries:         00000080
gpt partition entry 00
        partition type guid:    D5F7127456A1134B81DC867174929325
        partition guid:         8C9CBCE7EA3BA54A96366416ACC57F55
        first lba:      0000000000000800
        last lba:       00000000000107FF
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 01
        partition type guid:    AF3DC60F838472478E793D69D8477DE4
        partition guid:         2A5D77892DB0FB4B99B6B75AEAC6CA69
        first lba:      0000000000010800
        last lba:       0000000003AFBFDE
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 02
        partition type guid:    00000000000000000000000000000000
        partition guid:         00000000000000000000000000000000
        first lba:      0000000000000000
        last lba:       0000000000000000
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
gpt partition entry 03
        partition type guid:    00000000000000000000000000000000
        partition guid:         00000000000000000000000000000000
        first lba:      0000000000000000
        last lba:       0000000000000000
        attributes:     0000000000000000
        name:   000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
copying boot image ................................ done!
bbl loader
initializing sm
Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD... 
SD command cmd0         response : 01
SD command cmd55        response : 01
SD command cmd41        response : 01
SD command cmd55        response : 01
SD command cmd41        response : 00
[... 354 lines cut ...]
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./hello.elf...(no debugging symbols found)...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0xffffffe00064c0e4 in ?? ()
(gdb) x/3w 0x40000000
0x40000000:	Ignoring packet error, continuing...
Reply contains invalid hex digit 116
(gdb) x/3w 0x80000000
Ignoring packet error, continuing...
0x80000000:	Reply contains invalid hex digit 116
(gdb) x/i 0x1000
   0x1000:	Ignoring packet error, continuing...
Reply contains invalid hex digit 116
(gdb) 

Why did it report an error " 0x40000000:	Ignoring packet error, continuing..." and "0x80000000:	Reply contains invalid hex digit 116 "?

niwis, 2021-03-17:
Hi @gzhan88, it looks like you are trying to access physical memory after attaching gdb to the core running on virtual memory. This errors, because `0x40000000` and `0x80000000` are not mapped to the virtual memory space. If, however, this is what you want, you need to remove the following lines from [ariane.cfg](https://github.com/openhwgroup/cva6/blob/master/fpga/ariane.cfg):
https://github.com/openhwgroup/cva6/blob/4abae602c57045a2c89147a2c9c2cc47eab8fc14/fpga/ariane.cfg#L38-L40
This will result in the debugger exclusively accessing physical addresses.
Best,
Nils

gzhan88, 2021-03-18:
Hi niwis,thank you very much for your prompt reply.The problem has already been settled.

closed 2021-03-18 by gzhan88

Wcm926, 2021-04-07:
hi，I can not log in to linux by default username root when starting the operating system. Nothing happens when I type root , and the log is same as you.