openhwfoundation/cva6 #634: How to evaluate a benchmark correctly?

noureddine-as, 2021-03-17:
Hello,

I have an application, and I would like to evaluate how many cycles each instruction takes.

For that, I built the Verilator simulator, and I also compiled the application to run on top of `pk`.
From the traces I got, I see that many cycles are spent in Debug Mode ( https://github.com/openhwgroup/cva6/issues/601 , https://github.com/openhwgroup/cva6/issues/157 ).

So to get more accurate statistics, I do the following:
- I only consider the cycles where the application is in User Mode and only the instructions belonging to a given address range.
- I modified `src/ariane.sv`, so I can report, for each executed instruction, the number of cycles spent on it, not just the cycle where it occurred.
- I sum up all the numbers of cycles for each instruction

So my questions are:
- Is this methodology correct to get consistent statistics? Should I include those cycles spent in Debug Mode?
- How is the memory (DRAM, L1D, L1I) accesses modeled? 
  It is mentioned [here](https://cva6.readthedocs.io/en/latest/intro.html?highlight=Two%20cycle%20memory%20access#scope-and-purpose) that the data cache (or DRAM) has a latency of 1 cycle whereas the instruction cache has a 3-cycle latency; however, I could not locate these parameters in the repository, can you please tell me where they are if they are parametrizable?

Thanks a lot!

Best regards,

niwis, 2021-03-17:
Hello @noureddine-as,

Regarding Question 1:
as described in the threads that you referred to, the front-end server periodically uses debug mode to communicate with pk and emulate system calls. Therefore, if you're only concerned with the execution time of your application, you are right to exclude these intervals. For the rest of your procedure, it might be easier and more accurate to use the cycle count given by the instruction traces and take the difference of back-to-back instructions.

Regarding Question 2:
These given access latencies are not parameters but follow directly from the design/implementation of the memory subsystem. Note that they assume TLB/Cache hits and will increase otherwise.

Best,
Nils

noureddine-as, 2021-03-17:
Thanks for the clear answers @niwis !

Best regards,
Noureddine

closed 2021-03-17 by noureddine-as