openhwfoundation/cva6 #636: Building verilator model fails because of invalid cast

tunefish777, 2021-03-19:
Hi!

I tried building a verilator model today and encountered the following during `make verilate`:

```
../tb/ariane_tb.cpp:323:49: error: invalid cast from type ‘VlUnpacked<long unsigned int, 33554432>’ to type ‘void*’
memif.read(0x80000000, mem_size, (void *)top->ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_ram__DOT__Mem_DP);
```

I have a fully functional riscv-toolchain and I am using verilator 4.201 on Debian 10.
The commands I used to get to this error were:
```
https://github.com/openhwgroup/cva6.git
git submodule update --init --recursive
make verilate
```

Would really appreciate some help.

monideepbora, 2021-04-14:
I am encountering the same issue. Tried an older checkout from February 2021 but the issue persists. Is there a fix ?

tunefish777, 2021-04-14:
> I am encountering the same issue. Tried an older checkout from February 2021 but the issue persists. Is there a fix ?

The issue can be prevented by using Verilator 4.102. I've tried a few versions and this seems to be the only one that actually works. Obviously this is just a workaround and not a fix.

monideepbora, 2021-04-14:
Thanks. I just tested with Verilator 4.200. Seems to work as well. I installed the version from the "stable" branch of the git repo of verilator.

NazerkeT, 2021-05-27:
Dear @tunefish777 @monideepbora, could you please share which versions of the riscv-gnu-toolchain, riscv-isa-sim and riscv-pk have you used? Or have you installed them from somewhere else? I have built the last versions of each and simulated Ariane with the different versions of the verilator(4.200, 4.104, 4.014). However, unfortunately, my simulation stalls right from the beginning. So, I am a bit curious about which version combinations are working correctly within a community.

Thanks,
Best regards,

tunefish777, 2021-05-27:
@NazerkeT Can you elaborate on what you mean by "the simulation stalls"? If building the Verilator model works for you then this is not the right place to ask for software-related issues.
I am using a custom version of the riscv-tools on which I cannot share more information, but it's based on GCC 8.2.0.

NazerkeT, 2021-05-27:
> @NazerkeT Can you elaborate on what you mean by "the simulation stalls"? If building the Verilator model works for you then this is not the right place to ask for software-related issues.
> I am using [...]

My apologies for the inaccurate wording. By "simulation stalls" I mean that `make verilate` stalls in one of the Ariane_testharness runs. Nevertheless, thanks for the information on riscv-tools cutom version. 

Thanks,
Best regards,

Moschn, 2021-05-27:
Maybe you just need to wait a bit: #589.

NazerkeT, 2021-05-27:
> Maybe you just need to wait a bit: [#589 (comment)](https://github.com/openhwgroup/cva6/issues/589#issuecomment-762712894).

Thanks for the pointer! Actually, after 15 minutes of run `make verilate` completed successfully. So, now I am sticking to the same situation as in the linked issue with elf simulation. I will move there to describe my run-time environment.

Thanks,
Regards,

MikeOpenHWGroup, 2023-02-17:
Hello @NazerkeT, @tunefish777, @Moschn and @monideepbora, thanks for your interest in CVA6.  There has not been a lot of comments on this for ~1.75 years, and it seems to be resolved so I am closing this issue.  Please feel free to open another if you are still having problems.

closed 2023-02-17 by MikeOpenHWGroup