openhwfoundation/cva6 #637: [Synth 8-439] module 'plic_regs' not found

zslwyuan, 2021-03-20:
Hi all,

    THANKS for your excellent work! !
    However, when I am trying to build the RISCV system project via Vivado and OpenPiton, I encounter an error as shown below:

      [Synth 8-439] module 'plic_regs' not found ["/home/tingyuan/Tests/openpiton/piton/design/chip/tile/ariane/src/rv_plic/rtl/plic_top.sv":86]
     
     I searched in the repository of CVA5 and OpenPiton, and I failed to the definition of the module 'plic_regs'. In the directory 'rv_plic', I only found the following design files:

     plic_regmap.sv  plic_top.sv  prim_subreg_ext.sv  prim_subreg.sv  pyversion  rv_plic_gateway.sv  rv_plic_reg_pkg.sv  rv_plic_reg_top.sv  rv_plic.sv  rv_plic_target.sv  tlul_pkg.sv  top_pkg.sv

    I failed to locate the problem and so seek your help!
    Thanks in advance!!! ^_^

zslwyuan, 2021-03-20:
I located the cause of the problem:
when the script setup.tcl calls the command: ./gen_plic_addrmap.py -t $num_targets -s $num_sources > plic_regmap.sv
it ends up with an fatal error: Fatal Python error: Py_Initialize: Unable to get the locale encoding

I manually run the command ( ./gen_plic_addrmap.py -t $num_targets -s $num_sources > plic_regmap.sv) in the terminal to generate the module and the problem solved. However, I am still confused  by the fatal error. ^_^/

MikeOpenHWGroup, 2023-02-17:
Hi @zslwyuan, thanks for your interest in CVA6.  This issue seems to be resolved and has not received a response in almost two years, so I am closing it now.

closed 2023-02-17 by MikeOpenHWGroup