openhwfoundation/cva6 #638: openOCD debugging on zynq7020 FPGA

leemango1998, 2021-03-23:
Hey! I find this project [cva6-softcore-contest](https://github.com/ThalesGroup/cva6-softcore-contest) ported cva6 into zybo-z7-20. And I am working to port cva6 into pynq-z2 (zynq7020), which has the same core as zybo-z7-20. I used BRAM as memory on fpga, so it's quite  easy to port. I followed the steps in Getting Started and everything went well until debug with OpenOCD.

After programming the zynq7020, when load OpenOCD,it shows like this:

```
 mango@ubuntu:~/vivado/pynq-z2-cva6$ openocd -f fpga/openocd_digilent_hs2.cfg
 Open On-Chip Debugger 0.10.0+dev-00198-g35eed36ff (2020-11-23-00:55)
 Licensed under GNU GPL v2
 For bug reports, read
   http://openocd.org/doc/doxygen/bugs.html
 none separate
 adapter speed: 1000 kHz
 Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
    TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
  0 riscv.cpu              Y     0x00000000 0x249511c3     5 0x01  0x03
 Info : clock speed 1000 kHz
 Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
 Info : datacount=2 progbufsize=8
 Info : Examined RISC-V core; found 1 harts
 Info :  hart 0: XLEN=32, misa=0x40141105
 Info : Listening on port 3333 for gdb connections
 Ready for Remote Connections
 Info : Listening on port 6666 for tcl connections
 Info : Listening on port 4444 for telnet connections
 Info : accepting 'gdb' connection on tcp/3333
 Error: Timed out after 120s waiting for busy to go low (abstractcs=0x8001002). Increase the timeout with riscv set_command_timeout_sec.
 Error: Abstract command ended in error 'busy' (abstractcs=0x8001102)
 Error: Timed out after 120s waiting for busy to go low (abstractcs=0x8001102). Increase the timeout with riscv set_command_timeout_sec.
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Warn : negative acknowledgment, but no packet pending
 Info : dropped 'gdb' connection**
```
 
In gdb:

```
mango@ubuntu:~/vivado/pynq-z2-cva6$ riscv32-unknown-elf-gdb sw/app/helloworld_printf.riscv
GNU gdb (GDB) 8.0.50.20170724-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later http://gnu.org/licenses/gpl.html
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv32-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
http://www.gnu.org/software/gdb/bugs/.
Find the GDB manual and other documentation resources online at:
http://www.gnu.org/software/gdb/documentation/.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from sw/app/helloworld_printf.riscv...done.
(gdb) target remote :3333
Remote debugging using :3333
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Remote 'g' packet reply is of odd length: E0E
(gdb)
```

I don't know what's gonging wrong and not familiar with openOCD. So any help about that would be greatly appreciated!

S-Nomii, 2021-03-24:
It seems that the hardware encounter some sequential problem(the openocd command retry interval is bigger than the command executing time, which will cause busy error every time). Normally Openocd are able to handle this error. But if your core and jtag frequency ratio is really low, you have to change openocd settings by giving the suggested cmd through telnet. 

Try riscv set_command_timeout_sec 5  on telnet

S-Nomii, 2021-03-24:
It seems that the hardware encounter some sequential problem(the openocd command retry interval is smaller than the command executing time, which will cause busy error every time). Normally Openocd are able to handle this error. But if your core and jtag frequency ratio is really low, you have to change openocd settings by giving the suggested cmd through telnet.

Try riscv set_command_timeout_sec 5 on telnet

niwis, 2021-03-24:
Hello @leemango1998,
Adding to that, it is possible that an abstract command throws an exception (e.g. due to a memory access fault), which would lead to the behavior that you posted. I would suggest to double-check your PMPs, and perhaps try an RTL simulation using remote bitbang to see what is going on.
Best,
Nils

leemango1998, 2021-03-25:
@S-Nomii  @niwis Thanks for your reply! I have tried to set "riscv set_command_timeout_sec 5", and it immediately shows these errors. I used the same cpu core and JTAG wire as  cva6-softcore-contest. So maybe these errors are about pmps, as niwis said. I am going to check that, thanks!

closed 2021-04-02 by leemango1998