openhwfoundation/cva6 #642: Question regarding WT DCACHE write buffer

zzhu35, 2021-03-26:
Hi all, my name is Zeran and I am implementing a cache hierarchy using CVA6. I am confused by these lines: https://github.com/openhwgroup/cva6/blob/master/src/cache_subsystem/wt_dcache.sv#L219-L221
Could someone give me more insight about this TODO? I think write buffer should be disabled because you are not flushing the WT_DCACHE on fences and amos with the rl annotation. If write buffers are enabled and not flushed on the fences and release points, how do you ensure the global ordering of memory accesses across these boundaries?

Thank you very much!

niwis, 2021-03-30:
Hi Zeran,
the write buffer is drained before executing AMOs and `fence`s.
AMOs:
https://github.com/openhwgroup/cva6/blob/fbe1e01e2090b19bcf49d9d8c828cceeb7daab12/src/cache_subsystem/wt_dcache_missunit.sv#L383-L388
`fence`s:
https://github.com/openhwgroup/cva6/blob/fbe1e01e2090b19bcf49d9d8c828cceeb7daab12/src/commit_stage.sv#L196
Best,
Nils

MikeOpenHWGroup, 2023-02-17:
Thanks for your insights @niwis!

Hi @zzhu35, since Nils has provided a response and you have not updated this issue in almost 2 years, I will assume this issue can be closed.

closed 2023-02-17 by MikeOpenHWGroup