openhwfoundation/cva6 #647: Coprocessor interface with the cva6 core

rtefdsf, 2021-04-09:
**i have a coprocessor i designed to do image processing as the image enters the cva6 core and by a software on it will be transfer the image to my coprocessor then do the processing which had been chosen by custom instruction in the core then give the image back to the core again to present the result of the processing ...

- so i want to ask how to make the software to run on the core?

- and the next question what is the suitable way to make the interface between co-processor and the core to send and get the data how to design this interace or bus??**

MikeOpenHWGroup, 2023-02-17:
These are big questions @rtefdsf!

> so i want to ask how to make the software to run on the core

My recommendation would be to create new instructions in the custom extension space of the RV ISA.  All custom instructions can be sent to the co-processor.  This means you do not need to modify the CVA6 pipeline.

> what is the suitable way to make the interface between co-processor and the core

The CVA6 now supports the [CV-X-IF](https://github.com/openhwgroup/core-v-xif) and you can create a co-processor for that interface.

closed 2023-02-17 by MikeOpenHWGroup

mrbilandi, 2024-09-11:
Hi @MikeOpenHWGroup 

I want to implement a new instruction, but I can't find a clear explanation of how to do it. Is there any guide on how to add a custom instruction to the toolchain (I'm not sure where) and how to build our program with the new compiler?

The CVA6 already has an example coprocessor with the instruction "adds rs1, rs2(, rs3)." If I want to use this instruction in my code, how can I do that and how can I compile the code containing this instruction?