openhwfoundation/cva6 #649: Mounting Linux image in QuestaSim Simulation

0ena, 2021-04-12:
Dear Dr. Zaruba and project contributors,

I am in the process of simulating Ariane with the Linux image mounted on it, in QuestaSim. From what I see in previous threads [[1]](https://github.com/openhwgroup/cva6/issues/6#issuecomment-367030272) [[2]](https://github.com/openhwgroup/cva6/issues/314) this is possible.

I have a couple of questions that I would greatly appreciate if someone can help me with:

a) You are mentioning a time of ~1.5 days [[3]](https://github.com/openhwgroup/cva6/issues/42) in order to mount the image in simulation. Is this time with preloading or not? From what I have seen, there is an option to preload the Linux image in the memory. Is `make sim preload=vmlinux` the correct command to preload the Linux image in the memory and simulate it (vmlinux image being inside the `./cva6` folder)? 

b) What do I expect to see inside the `transcript` file that is generated? Do I expect to see all the booting sequence that I witness when booting in Linux using the FPGA? 
Until now, I have run the above mentioned command and I got the following inside the `transcript` file after ~2 days of running:
```
# ----------------------------------------------------------------
# UVM-1.1d
# (C) 2007-2013 Mentor Graphics Corporation
# (C) 2007-2013 Cadence Design Systems, Inc.
# (C) 2006-2013 Synopsys, Inc.
# (C) 2011-2013 Cypress Semiconductor Corp.
# ----------------------------------------------------------------
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_RXFF
# ** Warning: NUMERIC_STD."=": metavalue detected, returning FALSE
#    Time: 0 ns  Iteration: 0  Instance: /ariane_tb/dut/i_ariane_peripherals/gen_uart/i_apb_uart/UART_TXFF
# UVM_INFO tb/ariane_tb.sv(129) @ 0: reporter [Core Test] Preloading ELF: vmlinux
# [TRACER] Output filename is: trace_hart_0.log
# UVM_INFO tb/ariane_tb.sv(138) @ 80: reporter [Core Test] Loading Address: 0000000000000000, Length: 0000000000019074
# UVM_INFO tb/ariane_tb.sv(138) @ 80: reporter [Core Test] Loading Address: 000000000001a000, Length: 0000000000b767ec
```

c) From your experience, is it possible to run such a simulation at a post-synthesis/post-implementation stage or is it only possible with the pre-synthesis RTL?

d) Is it possible in simulation to run a simple C program when the Linux image has completely booted?

I would greatly appreciate any response on the above questions as it is the first time trying to go through such a complex simulation mechanism.

Thank you in advance for your responses and time!

Kind regards,
Nassos

zarubaf, 2021-04-13:
> a) You are mentioning a time of ~1.5 days [[3]](https://github.com/openhwgroup/cva6/issues/42) in order to mount the image in simulation. Is this time with preloading or not? From what I have seen, [...]

You need to boot the Linux image with the embedded bootloader. It is called `bbl` if you generate the Linux image using the `ariane-sdk`. The `vmlinux` alone runs with virtual memory (as the name suggests :-)) and you need the bootloader to prepare the environment. 

The `make sim preload=bbl` is the right command. @JeanRochCoulon also noted that they are booting Linux way faster on Verilator. He might have some wise words for you :-)

> b) What do I expect to see inside the `transcript` file that is generated? Do I expect to see all the booting sequence that I witness when booting in Linux using the FPGA?

Yes, the transcript should show you the entire boot process, pretty similar to the FPGA. The first output should be there after half an hour.

> c) From your experience, is it possible to run such a simulation at a post-synthesis/post-implementation stage or is it only possible with the pre-synthesis RTL?

While it technically could and should work (I don't see why not), it will take even longer so I think it isn't really feasible. What are you trying to achieve with booting it post-layout?

> d) Is it possible in simulation to run a simple C program when the Linux image has completely booted?

Technically yes, although you will need to make sure that they run without any user interaction. You won't have an interface to Linux as the character read isn't working in simulation. So you'd need to make sure that the user program runs as part of your init process (I've personally never done that). Alternatively, `pk` is the easy solution to run user programs without much intervention.

0ena, 2021-04-19:
Dear Dr. Zaruba,

> You need to boot the Linux image with the embedded bootloader. It is called `bbl` if you generate the Linux image using the `ariane-sdk`. The `vmlinux` alone runs with virtual memory (as the name [...]
Indeed the above command did the trick and I am able to witness the booting process in QuestaSim. It took about ~3 days to gather a .wlf of ~1.6TB that logged all the signals until around ~34s of the booting process.
I am more familiar with QuestaSim and it abides to industry standards (although way slower than the Verilator simulation). But I'll check the faster alternative of simulating and generating VCD files through Verilator (and then probably open them in Questa for inspection). Thank you for your comments.

> Yes, the transcript should show you the entire boot process, pretty similar to the FPGA. The first output should be there after half an hour.
>
>
> While it technically could and should work (I don't [...]
I am making some "minimally invasive" modifications of combinational nature in specific modules and I would like to see how they affect the booting process, so a post-implementation simulation would be useful in this regard. I am thinking of generating VCDs of signals from specific modules from the original and the modified design and compare them to see if they match. Because at the moment when I am booting Linux in the modified design, the core hangs. So I think that would be a good way to debug my problem.

> Technically yes, although you will need to make sure that they run without any user interaction. You won't have an interface to Linux as the character read isn't working in simulation. So you'd need [...]

Thank you for the information on this one! I will examine the option of the init-process, (pk if I recall correctly runs programs in Machine mode, so it doesn't fit my needs at the moment).

I really appreciate your responses.

Kind regards,
Nassos

closed 2021-04-21 by 0ena