openhwfoundation/cva6 #654: code point 0x120000f3 with rd=1 retires as sfence.vma

tzwaenn, 2021-04-19:
This is a collateral finding on the RTL from OneSpin processor app formal analysis.
According to the RISC-V privileged ISA, only "0001001 rs2 rs1 000 00000 1110011" should decode as sfence.vma (Sail has the decoding as 0b0001001 @ rs2 @ rs1 @ 0b000 @ 0b00000 @ 0b1110011). However, cva6 seems to ignore the 5 bits of the RD fiels leading to, for example, 0x120000f3 being decoded as sfence.vma with rd=1 and actually retiring as no exception flag is set in the scoreboard and thus, it is flushed in conjunction with the instruction retired counter incrementing.
Expected behavior would be to flag this as illegal.
[image]
See attached waveform - the problematic instruction word is fetched at -3, issued at -2, makes it to slot 0 of the scoreboard at -1, the scoreboard entry gets marked as valid at -1 and in this chycle, the scoreboard is flushed while the sfence commits and leads to the instruction retired counter being incremented from 0 to 1.
Note that I cut the 8-bit counters in the cashes to get to a quick counterexample, but that should not be related to the reported issue which looks like a decoder issue. Git hash of examined RTL is 6000e32.

MikeOpenHWGroup, 2023-02-17:
Hi @zarubaf and @JeanRochCoulon.  This is an old issue, but it comes from OneSpin (@tzwaenn) and I have come to greatly respect the work of this team.  I am willing to bet a couple of beer that this is a real bug.

JeanRochCoulon, 2023-03-16:
@AyoubJalali  As you can see, a bug has been found on SFENCE.VMA instruction. Is this instruction included in ISACOV ? What could be planned to cover it by the verification ?

AyoubJalali, 2023-03-16:
@tzwaenn Hello, thanks for the issue, do you have the source file S or something to regenerate the bug

AyoubJalali, 2023-03-16:
@JeanRochCoulon No does not exist in the ISACOV

tzwaenn, 2023-03-16:
Hi,
I do not have my setup from 2 years ago anymore. I suggest trying instruction 0x120000f3 in the decoder. If it raises an illegal exception, the issue has been fixed in the meantime. If it gets decoded as sfence.vma, you have the problem reproduced.

ASintzoff, 2023-10-20:
Fixed by commit b6c1d04b
in PR https://github.com/openhwgroup/cva6/pull/933

closed 2023-10-20 by ASintzoff