openhwfoundation/cva6 #656: Crash a Bare metal test CVA6-RISC-V assembly hello world

Bill94l, 2021-04-21:
After implementing cva6 on the VCU118 FPGA board, I did a baremetal test to display "hello world". For the compilation I used RISC-V GNU Compiler Toolchain ..... to load the binary and start its execution I used openOCD and GDB.

The program crashes ....... After the step-by-step display of the executed assembler instructions, the processor only executes the first five assembler instructions,  **This crash can be caused by what ?** 

I dont understand why the **PC** of the third instruction is **not aligned**(0x8000008**a**)?

## Section assembly 
```.D
Déassemblage de la section .init :

0000000080000080 <_start>:
    80000080:	00004197          	auipc	gp,0x4
    80000084:	9e818193          	addi	gp,gp,-1560 # 80003a68 <__global_pointer$>
    80000088:	00020117          	auipc	sp,0x20
    8000008c:	f7810113          	addi	sp,sp,-136 # 80020000 <__heap_end>
    80000090:	00000517          	auipc	a0,0x0
    80000094:	f7050513          	addi	a0,a0,-144 # 80000000 <__vector_start>
    80000098:	00156513          	ori	a0,a0,1
    8000009c:	30551073          	csrw	mtvec,a0
    800000a0:	7b818513          	addi	a0,gp,1976 # 80004220 <g_stdio_uart_init_done>
    800000a4:	00004617          	auipc	a2,0x4
    800000a8:	1c460613          	addi	a2,a2,452 # 80004268 <__bss_end>
    800000ac:	8e09                	sub	a2,a2,a0
    800000ae:	4581                	li	a1,0
    800000b0:	4a3000ef          	jal	ra,80000d52 <memset>
    800000b4:	00001517          	auipc	a0,0x1
    800000b8:	bfc50513          	addi	a0,a0,-1028 # 80000cb0 <__libc_fini_array>
    800000bc:	3cd000ef          	jal	ra,80000c88 <atexit>
    800000c0:	429000ef          	jal	ra,80000ce8 <__libc_init_array>
    800000c4:	4501                	li	a0,0
    800000c6:	4581                	li	a1,0
    800000c8:	4601                	li	a2,0
    800000ca:	00a000ef          	jal	ra,800000d4 <main>
    800000ce:	3c50006f          	j	80000c92 <exit>

00000000800000d2 <_fini>:
    800000d2:	8082                	ret
```
## Step-by-step display of the executed assembler instructions
```shell
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x000000000001003c in ?? ()
------------------------------------------------------------------------------------------------
(gdb) load
Loading section .vectors, size 0x80 lma 0x80000000
Loading section .init, size 0x54 lma 0x80000080
Loading section .text, size 0x2ee8 lma 0x800000d4
Loading section .rodata, size 0x27a lma 0x80002fc0
Loading section .init_array, size 0x8 lma 0x80003240
Loading section .data, size 0xf98 lma 0x80003248
Loading section .sdata, size 0x40 lma 0x800041e0
Start address 0x0000000080000080, load size 16918
Transfer rate: 45 KB/sec, 2416 bytes/write.
---------------------------------------------------------------------------------------------------
(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xffffffffffffff70	-144
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x80000080	0x80000080 <_start>
dscratch       0x0	0
---------------------------------------------------------------------------------------------------
(gdb) si
0x0000000080000084 in _start ()

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80004080	0x80004080 <__malloc_av_+1712>
..
a0             0xffffffffffffff70	-144
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x80000084	0x80000084 <_start+4>
dscratch       0x0	0
---------------------------------------------------------------------------------------------------
(gdb) si
0x0000000080000088 in _start ()

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xffffffffffffff70	-144
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x80000088	0x80000088 <_start+8>
dscratch       0x0	0
-------------------------------------------not aligned--------------------------------------------------------
(gdb) si 
**0x000000008000008a in _start ()**

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xffffffffffffff70	-144
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x8000008a	0x8000008a <_start+10>
dscratch       0x0	0
---------------------------------------------------------------------------------------------------
(gdb) si
0x0000000080000094 in _start ()

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xffffffffffffff70	-144
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x80000094	0x80000094 <_start+20>
dscratch       0x0	0
---------------------------------------------------------------------------------------------------
(gdb) si
0x0000000080000098 in _start ()

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xfffffffffffffee0	-288
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
..
pc             0x80000098	0x80000098 <_start+24>
dscratch       0x0	0
--------------------------------------------CRASH-------------------------------------------------------
(gdb) si
0x0000000000010044 in ?? ()

(gdb) info reg
ra             0x8000008e	0x8000008e <_start+14>
sp             0x83fffff0	0x83fffff0
gp             0x80003a68	0x80003a68 <__malloc_av_+152>
..
a0             0xfffffffffffffee0	-288
a1             0x1c2000	1843200
a2             0x0	0
a3             0x0	0
a4             0x10000000	268435456
a5             0x20	32
a6             0x0	0
a7             0x0	0
s2             0x80	128
..
pc             0x10044	0x10044
dscratch       0x0	0
(gdb) 
```

niwis, 2021-04-21:
Hello @Bill94, to understand the cause for the crash, could you read the CSRs `mcause` and `mtval` after the core jumps to the trap vector in ROM? Since the program flow seems not to be aligned with your elf file, could you also read the memory at your program counter (e.g. `disp /4i $pc`)?

Bill94l, 2021-04-21:
Hello @niwis, 
Thank you for your quick response.
Here I display the values of the different registers  *mcause* and *mtval* with the contents of the memory in different *PC*.

```Shell
.
..
----------------------------------------------------------------------------------------
(gdb) si
0x0000000080000088 in _start ()

1: x/4i $pc
=> 0x80000088 <_start+8>:	auipc	sp,0x20
   0x8000008c <_start+12>:	addi	sp,sp,-136
   0x80000090 <_start+16>:	auipc	a0,0x0
   0x80000094 <_start+20>:	addi	a0,a0,-144

(gdb) p/x $mcause 
$5 = 0x2

(gdb) p/x $mtval
$6 = 0x0
-------------------------------------------------------------------------------
(gdb) si
0x000000008000008a in _start ()

1: x/4i $pc
=> 0x8000008a <_start+10>:	c.slli64	zero
   0x8000008c <_start+12>:	addi	sp,sp,-136
   0x80000090 <_start+16>:	auipc	a0,0x0
   0x80000094 <_start+20>:	addi	a0,a0,-144

(gdb) p/x $mcause 
$7 = 0x2

(gdb) p/x $mtval
$8 = 0x0
---------------------------------------------------------------------------
(gdb) si
0x0000000080000094 in _start ()

1: x/4i $pc
=> 0x80000094 <_start+20>:	addi	a0,a0,-144
   0x80000098 <_start+24>:	unimp
   0x8000009a <_start+26>:	addi	a0,a0,338
   0x8000009e <_start+30>:	sd	ra,24(sp)

(gdb) p/x $mcause 
$9 = 0x2

(gdb) p/x $mtval
$10 = 0x0
-----------------------------------------------------------------------------
(gdb) si
..
.
-----------------------------------------------------------------------------
(gdb) si
0x0000000080000098 in _start ()

1: x/4i $pc
=> 0x80000098 <_start+24>:	unimp
   0x8000009a <_start+26>:	addi	a0,a0,338
   0x8000009e <_start+30>:	sd	ra,24(sp)
   0x800000a0 <_start+32>:	addi	a0,gp,1976

(gdb) p/x $mcause 
$11 = 0x2

(gdb) p/x $mtval
$12 = 0x0
-----------------------------------------------------------------------------
(gdb) si
0x0000000000010044 in ?? ()

1: x/4i $pc
=> 0x10044:	andi	s2,s2,8
   0x10048:	beqz	s2,0x1003c
   0x1004c:	lui	s1,0x2000
   0x10050:	csrr	s2,mhartid

(gdb) p/x $mcause 
$11 = 0x2

(gdb) p/x $mtval
$12 = 0x0
```

niwis, 2021-04-21:
Did you try to execute the binary in simulation? (`make sim preload=file.elf`)

Bill94l, 2021-04-21:
This is result of the  binary execution in simulation
```Shell
$ work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk   helloworld_printf.elf 

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 37983

terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #9512165362959908865
```

niwis, 2021-04-21:
If your test program is bare-metal (i.e. mapped to physical addresses, `0x80000000` upwards), you probably do not want the proxy kernel. Change the command to `work-ver/Variane_testharness helloworld_printf.elf`

Bill94l, 2021-04-22:
With this command  `work-ver/Variane_testharness helloworld_printf.elf` a warning is displayed and the simulation is blocked
```Shell
$ work-ver/Variane_testharness  helloworld_printf.elf 

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38779

warning: tohost and fromhost symbols not in ELF; can't communicate with target

```

Bill94l, 2021-09-05:
The problem is I did not activate the "NARROW BURST" option when configuring the DDR4 (DataWidth: 64bit / AxiDataWidth: 64 * 8 = 512 bit). which is adapted in our case because the size of the cache line (128bit) is smaller than the size of the ddr4 output frame.

closed 2021-09-05 by Bill94l

yulong-lan, 2024-03-09:
Hi @Bill94l , did you happen to encounter this issue ? #1912 #