openhwfoundation/cva6 #658: Ethernet constraints

abdullahyildiz, 2021-04-25:
I noticed that Ethernet pins do not have min and max delay constraints in [genesys-2.xdc](https://github.com/openhwgroup/cva6/blob/6000e32b4181304901adc4c8d72870ebc94473a5/fpga/constraints/genesys-2.xdc) to meet RGMII timing. Is there a specific reason for that?

abdullahyildiz, 2021-04-25:
According to the [board schematic](https://reference.digilentinc.com/_media/reference/programmable-logic/genesys-2/genesys-2_sch.pdf) and [reference manual](https://reference.digilentinc.com/reference/programmable-logic/genesys-2/reference-manual#ethernet_phy) of Genesys 2 board:

- The PCB traces do not introduce the required skews.
- PHY is configured to introduce the delay on the FPGA ETH_RX* interface, by delaying its ETH_RX_CLK output by 2 ns.
- PHY is not configured to introduce a delay on the PHY ETH_TX* interface, hence Ethernet MAC should output a [clock](https://github.com/lowRISC/ariane-ethernet/blob/6a5436bf110f83ebb13119dbd82650ccd8f947c9/rgmii_phy_if.sv#L209) with 90 degree phase shift.

Considering the above points, can we conclude that there is no need to set I/O delay constrains for Ethernet?

zarubaf, 2021-04-27:
Good question. I haven't looked very deeply into it. The ethernet is derived from [Alex Forencich's ethernet](https://github.com/alexforencich/verilog-ethernet) afaik and was adapted and contributed from lowRISC (@jrrk). Maybe they have some more input?

jrrk2, 2021-04-27:
This is a good review of the situation, additional constraints should not be necessary when I/O flip flops have been used, which is desirable at this frequency. This adaptation was produced some time ago without too much thought. Are you looking for causes of unreliability at the moment? One module which was not brought across from LowRISC was the block which reads the MAC address from the OTP area, this was another hack, which would be quite useful to implement in Ariane.

Jbalkind, 2021-05-04:
You might also like to look at the constraints we set for openpiton+ariane on genesys2 (though it's a different mac) https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/design/xilinx/genesys2/constraints.xdc

Also the various special I/O pins: https://github.com/PrincetonUniversity/openpiton/blob/openpiton/piton/design/chipset/rtl/chipset.v#L1443-L1528

Do note however that we were using something like an MII mac and transducing to RGMII at that interface so there is a little weirdness due to that.

abdullahyildiz, 2021-05-04:
@jrrk2,

I just wanted to understand the logic behind it.
By the way, I couldn't find any IOB property that is applied to ethernet ports (either in XDC and RTL code).

jrrk2, 2021-05-04:
I’m not sure that IOB properties are used in Vivado, you need to check if they are just an ISE thing.

MikeOpenHWGroup, 2023-02-17:
Thanks for yours insights @jrrk2, @Jbalkind and @zarubaf.

Hi @abdullahyildiz, given that this issue has not received any further comments in ~1.75 years, I will close it now.

closed 2023-02-17 by MikeOpenHWGroup