openhwfoundation/cva6 #666: Data cache implementation

abdoboua, 2021-05-10:
I don't understand the meaning of DCACHE_NUM_WORDS , I did some calculations based on the standard configuration of the Dcache and I found that it is a  8 way set-associative cache, each set contains 8 Cache Lines with each line is 128 bits. Each CL is separated into to words of 64bits stored in two different banks ( bank0 contains word0 of each CL ,   bank1 contains word1 of each CL) .But when I calculated DCACHE_NUM_WORDS I found (2^8=256) . Moreover, Is the Dcache write through? because I found in the description of some src files (wt_dcache_wbuffer and wt_dcache_mem) that the used blocks are for wb cache .

Another question: Is there any available detailed documentation to understand the microarchitecture of cva6 (implemented protocols and design choices) ?

E-Poulin, 2021-05-24:
Hello, 
From what I have understood, DCACHE_NUM_WORDS is the number of sets of the D$. For the standard configuration there are 256 sets, with 8 CL in each, and each line is 128 bits. Then the offset_width is 4 bits (log2(Line_width/8)).

You can chose between WT and WB dcache if you define (or not) WT_DCACHE. If WT_DCACHE is defined in the Makefile, it will use wt_dcache_subsytem and if not, it will use std_cache_subsystem (aka WB cache).
https://github.com/openhwgroup/cva6/blob/6963fe9667d68c06170940f01bcb0ed61e816074/src/ariane.sv#L607-L688

Best,
Emeric

MikeOpenHWGroup, 2023-02-17:
Thanks for your response @E-Poulin.

Hi @abdoboua, given that there has not been a further comment on this Issue for > 1.5 years, I am closing it.