openhwfoundation/cva6 #672: make verilate command includes problematic axi2apb_wrap module

MikePopoloski, 2021-05-31:
The sv-tests repo uses the makefile's verilate command in order to get the file list to test Ariane; see here for context: https://github.com/SymbiFlow/sv-tests/pull/1565

The problem we're seeing is that the file list includes axi2apb_wrap.sv, which references an unknown interface APB_BUS in its port list. It seems like the axi2apb_wrap module itself isn't used elsewhere in this project, so perhaps it should be removed from the file list? I definitely could be missing something though.

MikeOpenHWGroup, 2023-02-17:
Hi @MikePopoloski, there have been no further comments for ~1.5 years. I believe this issue is resolved, and can be closed.  If that is not the case, please feel free to reopen this issue or create a new one.

closed 2023-02-17 by MikeOpenHWGroup

MikePopoloski, 2023-02-17:
I seem to be unable to reopen this, but the issue is not resolved. The Ariane build used by sv-tests is just generally broken all the time (as it is currently). If someone on this project could take a look and suggest something more resilient, or if they could make the `make verilate` command actually functional, that would be appreciated.