openhwfoundation/cva6 #674: schema : realigner at the wrong place

ld3804, 2021-05-31:
I think the schema ariane_overview.png (cva6/docs/_static/ariane_overview.png) is wrong : the realigner has to be in frontend stage before instruction queue, and not in decode stage after the instruction queue.

[image]

Bill94l, 2021-06-02:
According to the CVA6 paper : https://arxiv.org/pdf/1904.05442.pdf

**Instruction Decode**: 
Re-aligns potentially unaligned instructions, de-compresses them and decodes them. Decoded instructions are then put into an issue queue in the issue stage.
**[source :  CVA6 Paper]**

ld3804, 2021-06-03:
> According to the CVA6 paper : https://arxiv.org/pdf/1904.05442.pdf
> 
> **Instruction Decode**:
> Re-aligns potentially unaligned instructions, de-compresses them and decodes them. Decoded instructions are then put into an issue queue in the issue stage.
> **[source : CVA6 Paper]**

This doc was edited before the 4.2.0 release, so i think this part is out of date because it's a change between Ariane 4.1.2 and Ariane 4.2.0

I checked the Ariane 4.1.2, and a "soft realigner" was on the frontend stage and the re-aligner module was after the instruction queue, in the id_stage (like on the schema). 
With Ariane 4.2.0, the re-aligner is now placed before the instruction queue, and no re-aligner was on the id_stage.

The associated commit : https://github.com/openhwgroup/cva6/commit/830540b7574f57c2bf155cf4d75cd1849549eb0b

For Ariane 2.0.0, here the source code : 
- in [frontend.sv](https://github.com/openhwgroup/cva6/blob/485662f0c9a315e084c5e18db49b4ea4a7503dfe/src/frontend/frontend.sv), the realigner is before the instruction queue
- in [id_stage.sv](https://github.com/openhwgroup/cva6/blob/485662f0c9a315e084c5e18db49b4ea4a7503dfe/src/id_stage.sv) no realigner is present in this module

[Doc](https://github.com/openhwgroup/cva6/blob/485662f0c9a315e084c5e18db49b4ea4a7503dfe/docs/id_stage.md) is also out of date (no realignment in id_stage) 

I will edit the issue to add the link to the commit.

MikeOpenHWGroup, 2023-02-17:
Hi @Bill94l  and @ld3804, there have been no further comments for ~1.5 years and the last responses are "thumbsup". Given that, I believe this issue is resolved, and can be closed.  If that is not the case, please feel free to reopen this issue or create a new one.