openhwfoundation/cva6 #675: Error generating bit file in VCU118

Fog-cake, 2021-06-03:
Hi，
  I found that you have the constraint file corresponding to vcu118, and I also have a development board. So I modified all the board types that I could find in the Makefile. But the following error still occurs. But it is feasible to use kc705.
  The error is as follows.
[image]
[vivado.log](https://github.com/openhwgroup/cva6/files/6590275/vivado.log)

Fog-cake, 2021-06-03:
I solved the problem after make clean, so I'm sorry to bother you.

Fog-cake, 2021-06-03:
I am trying to specify the version of DDR3 to see how it turns out.

Fog-cake, 2021-06-03:
This may be a version bug of vivado2018.3?

Fog-cake, 2021-06-03:
The name mig_7series seems to be wrong.

Bill94l, 2021-06-04:
For VCU118 FPGA board, you need to use DDR4

Fog-cake, 2021-06-04:
> For VCU118 FPGA board, you need to use DDR4

Hi，Bill94l
Yes, I have switched to DDR4 and succeeded. But there is a problem, if I introduce DDR4_ sdram_ C1, in fact, can not output 200MHz clock. So I deleted this configuration.

Bill94l, 2021-06-04:
The 200Mhz is just for Ethernet. You will be able to generate 250Mhz which will feed the clock generator, then from the clock generator you can add an output which gives you the 200MHz

Bill94l, 2021-06-04:
did you manage to read the data from ddr4? I have a problem reading the data correctly

Fog-cake, 2021-06-04:
> did you manage to read the data from ddr4? I have a problem reading the data correctly

I see the question you asked in issue, but it's a pity that I haven't done that yet. My JTAG actually works through a transfer card, and I have to modify my constraint file again.

closed 2021-06-04 by Fog-cake