openhwfoundation/cva6 #679: make verilate issues

sebastien-riou, 2021-06-14:
I get: 
````
Variane_testharness__Trace__2__Slow.cpp: In static member function ‘static void Variane_testharness::traceInitSub9(void*, VerilatedVcd*, const char*)’:
Variane_testharness__Trace__2__Slow.cpp:3218:71: error: invalid conversion from ‘const char*’ to ‘int’ [-fpermissive]
 3218 |         tracep->declBus(c+24390,Verilated::catName(scopep,"BAUD_RATE"," "), false,-1, 31,0);
      |                                                                       ^~~
      |                                                                       |

````

Steps to reproduce on Ubuntu 20.04:
- verilator version: use v4.200 (v4.202 is NOT OK, other issues)
- make sure $RISCV/include contains fesvr folder, if you don't have it, get it from [here](https://github.com/riscv/riscv-isa-sim/tree/master/fesvr) 
- `make verilate DEBUG=1`

sebastien-riou, 2021-08-17:
seems like a verilator issue actually. I opened an issue there: https://github.com/verilator/verilator/issues/3101

MikeOpenHWGroup, 2023-02-17:
Hi @sebastien-riou, based on your comment and the fact that there have been no further comments for ~1.5 years, I believe this issue is resolved, and can be closed.

closed 2023-02-17 by MikeOpenHWGroup