openhwfoundation/cva6 #681: Kernel crash when accessing address in kernel address space from user process

0ena, 2021-06-14:
Hi,

I have the following question regarding the Linux running on Ariane.
When a user process accidentally tries to access an address inside the kernel address space (allocated or not to an LKM), the kernel crashes displaying the following message:

```
[   62.181919] user_process1.o[113]: unhandled signal 11 code 0x1 at 0xffffffd00400b400
[   62.181919] user_process1.o[113]: unhandled signal 11 code 0x1 at 0xffffffd00400b400
[   62.193285] CPU: 0 PID: 113 Comm: user_process1.o Tainted: G           O      5.1.0-rc7 #12
[   62.193285] CPU: 0 PID: 113 Comm: user_process1.o Tainted: G           O      5.1.0-rc7 #12
[   62.210756] sepc: 00000000000101da ra : 00000000000101cc sp : 0000003fff9b2d60
[   62.210756] sepc: 00000000000101da ra : 00000000000101cc sp : 0000003fff9b2d60
[   62.225360]  gp : 0000000000022970 tp : 0000001555752ec0 t0 : 00000000000102ec
[   62.225360]  gp : 0000000000022970 tp : 0000001555752ec0 t0 : 00000000000102ec
[   62.240059]  t1 : 0000000000001000 t2 : ffffffffffffffff s0 : 0000003fff9b2d80
[   62.240059]  t1 : 0000000000001000 t2 : ffffffffffffffff s0 : 0000003fff9b2d80
[   62.255185]  s1 : 00000000000a1600 a0 : 0000000000000001 a1 : 0000003fff9b1920
[   62.255185]  s1 : 00000000000a1600 a0 : 0000000000000001 a1 : 0000003fff9b1920
[   62.269881]  a2 : 0000000000000000 a3 : 0000003fff9b17f8 a4 : 00000000aaaabbbb
[   62.269881]  a2 : 0000000000000000 a3 : 0000003fff9b17f8 a4 : 00000000aaaabbbb
[   62.284808]  a5 : ffffffd00400b400 a6 : 0000003fff9b1920 a7 : ffffffffffffff9f
[   62.284808]  a5 : ffffffd00400b400 a6 : 0000003fff9b1920 a7 : ffffffffffffff9f
[   62.299712]  s2 : 000000000009a6bb s3 : 000000000008ab20 s4 : 00000000000a1640
[   62.299712]  s2 : 000000000009a6bb s3 : 000000000008ab20 s4 : 00000000000a1640
[   62.314628]  s5 : 0000000000000008 s6 : 0000001555752ed0 s7 : 000000000000000a
[   62.314628]  s5 : 0000000000000008 s6 : 0000001555752ed0 s7 : 000000000000000a
[   62.329533]  s8 : 0000000000000000 s9 : 0000000000089be8 s10: 00000000000a1600
[   62.329533]  s8 : 0000000000000000 s9 : 0000000000089be8 s10: 00000000000a1600
[   62.344442]  s11: 00000000000a3045 t3 : 0fffffffffffffff t4 : 0000000000000019
[   62.344442]  s11: 00000000000a3045 t3 : 0fffffffffffffff t4 : 0000000000000019
[   62.359487]  t5 : 0000000000000009 t6 : ffffffffffffffbf
[   62.359487]  t5 : 0000000000000009 t6 : ffffffffffffffbf
[   62.371406] sstatus: 8000000200006020 sbadaddr: ffffffd00400b400 scause: 000000000000000f
[   62.371406] sstatus: 8000000200006020 sbadaddr: ffffffd00400b400 scause: 000000000000000f
```

In the above example the user process accessed the kernel space address `0xffffffd00400b400`, which caused the kernel to crash with an error code `0xf` that corresponds to a `STORE_PAGE_FAULT`. It is obvious that a user process should not have access to the kernel address space, however from my knowledge I was expecting for the program/user process to crash instead of the whole kernel.

Can I safely assume that this behavior is attributed to the Linux image and is the expected?

Thank you in advance for your responses and time.

Kind regards,
Nassos

zarubaf, 2021-06-15:
Yeah that is interesting. I would say this is most probably related to the Linux kernel as the HW seems to be aware that this isn’t an allowed access hence, the page fault. Would be interesting to try this on Qemu or Spike and then ask somebody how knows more about the Kernel. Behavior doesn’t seem correct to me - it is also an oldish kernel though (in terms of RISC-V support).

MikeOpenHWGroup, 2023-02-17:
Thanks for the reply @zarubaf!

Hi @0ena, you have responded with a "thumbs-up" to Florian's reponse and there have been no further comments in > 1.5 years, so I assume this issue can be closed.  If the problem persists, please feel free to either re-open this issue or create a new one.

closed 2023-02-17 by MikeOpenHWGroup