openhwfoundation/cva6 #684: work-ver/Variane_testharness

LIUQyou, 2021-06-17:
hello,

I encounter a problem when running Variane_testharness, I used the Verilator 4.104 and riscv64-unknown-elf-gcc (GCC) 9.2.0.

I followed the instructions
```shell
echo '
#include <stdio.h>

int main(int argc, char const *argv[]) {
    printf("Hello CVA6!\\n");
    return 0;
}' > hello.c
riscv64-unknown-elf-gcc hello.c -o hello.elf
```
Then 
```shell
sudo work-ver/Variane_testharness -v hello.vcd $RISCV/riscv64-unknown-elf/bin/pk hello.elf
```
But it gives no reaction.
Then I checked the .vcd file. It shows like below:
[image]
it seems always running.

Could you give me some hints on this problem.

Thanks

sebastien-riou, 2021-08-17:
Running it without generating the VCD is long but it does complete:
````
> work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk mysoft/hello.elf
    This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
    Listening on port 39917
    bbl loader
    Hello CVA6!\n/opt/riscv/riscv64-unknown-elf/bin/pk completed after 2633204 cycles
    CPU time used: 267321.67 ms
    Wall clock time passed: 267662.70 ms
````

so you need to simulate 2.6 million cycles !

MikeOpenHWGroup, 2023-02-17:
Thanks for your response @sebastien-riou!

Hi @LIUQyou, this issue has not been updated in 1.5 years so I assume your problem has been resolved and will close the issue.  If that is not the case, feel free to re-open this issue or create a new one.

closed 2023-02-17 by MikeOpenHWGroup