openhwfoundation/cva6 #697: PMP and G value

masgia, 2021-07-26:
From the specs:
`In general, the PMP grain is 2 G+2 bytes and must be the same across all PMP regions. When G ≥ 1, the NA4 mode is not selectable. When G ≥ 2 and pmpcfg i .A[1] is set, i.e. the mode is NAPOT, then bits pmpaddr i [G-2:0] read as all ones. When G ≥ 1 and pmpcfg i .A[1] is clear, i.e. the mode is OFF or TOR, then bits pmpaddr i [G-1:0] read as all zeros.`

While in csr_registers.sv
```
// PMPADDR
// Important: we only support granularity 8 bytes (G=2)
// -> last bit of pmpaddr must be set 0/1 based on the mode:
// NA4, NAPOT: 1
// TOR, OFF:   0
riscv::CSR_PMPADDR0:         csr_rdata = {10'b0, pmpaddr_q[0][riscv::PLEN-3:1], (pmpcfg_q[0].addr_mode[1] == 1'b1 ? 1'b1 : 1'b0)};
```
8-bytes granularity would correspond to G=1 in my understanding.
Moreover, if the code is consistent with the rule which states that pmpaddr[G-2:0] read as all ones, it is not compliant with the one which states that pmpaddr[G-1:0] read as all zeros (it should be the last 2 bits, not the last one).

Also, it looks like NA4 is selectable.
I can write pmpcfg.a=NA4, and the PMP implementation behaves accordingly.
For example, with the following configuration
pmpcfg0.r = 1;
pmpcfg0.w = 1;
pmpcfg0.x = 0;
pmpcfg0.a = 2; // NA4
pmpcfg0.o = 0;
pmpcfg0.l = 0;
pmpaddr0=0x40000000;
I can read and write address 0x40000000 but I get an exception when accessing address 0x40000004.

masgia, 2021-07-26:
To me the easiest solution would be to support 4-bytes granularity (G=0) and change
`riscv::CSR_PMPADDR0:         csr_rdata = {10'b0, pmpaddr_q[0][riscv::PLEN-3:0]};`

Moschn, 2021-07-27:
> 8-bytes granularity would correspond to G=1 in my understanding.
> Moreover, if the code is consistent with the rule which states that pmpaddr[G-2:0] read as all ones, it is not compliant with the one which states that pmpaddr[G-1:0] read as all zeros (it should be the last 2 bits, not the last one).

I think you are partially correct. G indeed seems to be 1 instead of 2. 

However, if you read the quote from the spec carefully, the last two bits only have to be set to 0 **if** G>=2. As noted above, G=1 so that means the last two bits do **not** have to set to 0. I had to go over the PMP spec multiple times in the past, and I am still not quite confident that I understand it perfectly. So please do correct me if I am wrong.

Now, why are we setting the last bit to 0? To be honest, I am not sure if I remember the reasoning, but I think it should be consistent with the spec. If NAPOT is selected with G=1 then the last bit of `pmpaddr` always has to be 0 (see table below). 

[image]

> To me the easiest solution would be to support 4-bytes granularity (G=0) and change
> `riscv::CSR_PMPADDR0: csr_rdata = {10'b0, pmpaddr_q[0][riscv::PLEN-3:0]};`

If I remember correctly, our reasoning behind only allowing a granularity of 8 bytes (G=1) was based on the memory accesses that the core performs: The core always performs aligned memory accesses of 8 bytes no matter the actual instruction (e.g., `lh` loading 2 bytes or `lb` loading 1 byte). If we allow 4 byte granularity, then there could be a problem if:
`0x4000000` -`0x4000004` is allowed (I.e., your example from above), and then the core performs an 8 byte read (i.e., `ld`) of address 0x4000000. I believe that in this case, the PMP entry would not throw an exception which is in violation of the spec.

masgia, 2021-07-27:
1. If you want to go for G=1, then you should then prevent pmpcfg.a=NA4 from being set (which is currently possible). And yes, as you wrote, fix the else path.
2. in case of byte-aligned memory access, like
```
  volatile uint8_t * sram = (uint8_t *)0x80000000;

  for (int i = 0; i < 16; i++) {
    *(sram+i) = i+1;
    if (*(sram+i) != i+1)
      return 0xff;
  }
```
the corresponding AXI transactions are targeting addresses 0x8000000, 0x8000001, 0x8000002... and use the strobe field, so they actually are single byte accesses.
I would then conclude that G=0 is a valid option. 
Or do you see any issue with this reasoning?

Moschn, 2021-07-27:
>     1. If you want to go for G=1, then you should then prevent pmpcfg.a=NA4 from being set (which is currently possible). And yes, as you wrote, fix the else path.

I agree. However, if software actually tries to detect the granularity, then it will notice that G=1 and could realize it should not use NA4. Nevertheless, It is probably better to not allow to set `pmpcfg` to NA4 it all together.

> 2. in case of byte-aligned memory access, like
>
>
> ```
>   volatile uint8_t * sram = (uint8_t *)0x80000000;
>
>   for (int i = 0; i < 16; i++) {
>     *(sram+i) = i+1;
>     if (*(sram+i) != i+1)
>       return [...]

Maybe I failed to communicate properly. I will try again:

The current PMP implementation only checks the first byte of the memory access. It does not consider the size of the access. For instance, it would perform the same check for an `ld` at 0x8000000 loading 8 bytes and an `lb` at the same address loading 1 byte. This leads to many cornercases that need to be investigated.

However since the core does not support misaligned loads/stores ([source](https://github.com/openhwgroup/cva6/blob/5fd9bec35498bf6c981c06f99005d48ff65c2539/src/load_store_unit.sv#L309)) many of these cornercases fall away. The core determines misaligned accesses as follows:
- load/store is 8 byte wide: last 3 bits are required to be 0. 
- load/store is 4 byte wide: last 2 bits are required to be 0. 
- load/store is 2 byte wide: last 1 bit is required to be 0. 

Now to the actual cornercases:
Assume 0x80000000 - 0x80000004 to be accessible using NA4 and everything else not accessible
- `ld r10, 0x80000000`: The core loads 8 bytes from address 0x80000000 into register r10. The PMP unit would check the first byte, i.e., 0x80000000, and match it. Result: the load would be successful even if the last 4 bytes are outside of the accessible region.
- `lw r10, 0x80000002`: The core loads 4 bytes from address 0x80000002 into register r10. The PMP unit would check the first byte, i.e., 0x80000002, and match it. Result: the load would be successful even if the last 2 bytes are outside of the accessible region. **Note**: This is actually not the case since CVA6 does not allow misaligned accesses.
- `lh r10, 0x80000003`: The core loads 2 bytes from address 0x80000003 into register r10. The PMP unit would check the first byte, i.e., 0x80000003, and match it. Result: the load would be successful even if the last byte is outside of the accessible region. **Note**: This is actually not the case since CVA6 does not allow misaligned accesses.

As you can see, only one cornercase remains, where an 8 byte access crosses a NA4 PMP boundary. For these reasons, we chose to go with G=1 as it eliminates the problems noted above. One could try to fix the problem by checking every single accessed byte, but that would add more complexity for a very small benefit (when is anyone using NA4 in a 64bit processor anyways?).

PR #698 (2021-07-29): pmp: Fix csr
- Add proper granularity (G=1)
- Disallow NA4
- Fix read of pmpaddr
- Bump riscv-tests to support pmp benchmark that detects granularity (https://github.com/riscv/riscv-tests/pull/295)
- Add PMP benchmark to CI

Fixes #538 
Fixes #697
closed 2021-07-29 by zarubaf