openhwfoundation/cva6 #700: Synthesis failed

Saqlinahamad, 2021-08-04:
I a running Arinae Core synthesis in lattice radiant tool.
I am having syntax error with respect to instr_tracer.sv file 

@E: CG285 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/util/instr_tracer.sv":48:4:48:9|Expecting statement
@E: CG428 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/util/instr_tracer.sv":81:20:81:20|Expecting =, <=, or arithmetic-assignment operator
@E: CG103 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/util/instr_tracer.sv":175:20:175:20|Expecting expression
@E: CG103 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/util/instr_tracer.sv":182:23:182:23|Expecting expression

zarubaf, 2021-08-04:
The instruction tracer isn’t synthesisable and not needed for implementation. You can safely ignore it in the file list

closed 2021-08-04 by zarubaf

Saqlinahamad, 2021-08-04:
okay 
I got another error with the scoreboard module "undefined rr_arb_tree module" & when i defined the synthesis is failing. Are all the "src" directory files are synthesizable or some are excluded in it?

Moschn, 2021-08-04:
The `rr_arb_tree` module should definitely be synthesizable. It seems like you did not download all the sources. Try to clone all submodules as well (see README).

There are multiple manifests that describe which source files are necessary for synthesis: `Bender.yml`, `Flist.ariane`, `ariane.core`, `src_files.yml`. Take a look at them as they also specify which files are exclusive for simulation or synthesis.

Saqlinahamad, 2021-08-04:
okay. Thanks

Saqlinahamad, 2021-08-05:
Hlo,
When I added the synthesis files from Bender.yml, I am getting the following errors 
@E: CG103 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/fpu/src/fpnew_fma.sv":66:38:66:45|Expecting expression
@E: CS187 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/fpu/src/fpnew_fma.sv":673:0:673:8|Expecting endmodule
@E: CG103 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/fpu/src/fpnew_opgroup_multifmt_slice.sv":139:35:139:42|Expecting expression
@E: CS187 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/fpu/src/fpnew_opgroup_multifmt_slice.sv":426:0:426:8|Expecting endmodule
@E: CG855 :"/home/saqlinahamad.m/Radiant/CVA6/git_subtrees/cva6/src/common_cells/src/lfsr.sv":22:7:22:10|Duplicate module/interface name lfsr
@E::Verilog compiler failed

Moschn, 2021-08-05:
The FPU uses some rather modern systemverilog features with which the tool you are using seems to have problems with. Not sure if there is an easy solution for that (you could try things like `sv2v`)...

Saqlinahamad, 2021-08-06:
It's not a right platform to ask question 
kindly once go through this
After so many warnings, I got this error: 

child killed: segmentation violation

Fail to run synthesis -f cva6_Implementation0_lattice.synproj -gui -msgset /home/saqlinahamad.m/Radiant/project_bin/riscv-fpga-dev/synth/script/promote.xml

zarubaf, 2021-08-06:
I am certain that CVA6 synthesizes with Vivado, DC, and Genus. I am not familiar with the Lattice tool, but I assume that this is a problem with the Synthesis tool - you might want to check with Lattice support. Of course, if there is anything we can change in the code to improve compatibility we are more than happy to accept a pull request.

Saqlinahamad, 2021-08-06:
okay Thanks.

Saqlinahamad, 2021-09-23:
Hi 
I had raised a issue to lattice semiconductor regarding synthesis, 
They said :
Are the only files in the script make the synthesis of cva6 core, As when they are are checking each .sv file, they can't able to locate some defines that are written over the RTL Code. 
So I am sending an attachment of script file, could you let me know these are only files required for synthesis.