openhwfoundation/cva6 #703: No response from interconnect when adding slave

Bill94l, 2021-08-09:
I would like to add a custom ip to the principal  bus. The IP contains a slave interface and a master interface.

First, I wanted to add a BRAM on the bus for a step by step integration of the IP. For that, I started by creating a new input on the bus to integrate the BRAM.

After several tests, I realized that as soon as I add a new entry in the bus and without even integrating the BRAM, nothing works: 
 - The debugger does not respond through jtag.
 - Linux does not boot.

 I replaced the GPIO IP with the BRAM (I used the GPIO memory segment, without adding a new entry on the main bus) and all worked perfectly (Linux/ debugger through JTAG). 

Here is what I modify to add a new entry on the bus, just after this modification which consists in adding a new memory segment (with or without adding the IP) nothing works:

**tb/ariane_soc_pkg.sv**
``` C
  typedef enum int unsigned {
    DRAM = 0,
    GPIO = 1,
    Ethernet = 2,
    SPI = 3,
    Timer = 4,
    UART = 5,
    PLIC = 6,
    CLINT = 7,
    ROM = 8,
    Debug = 9,
    BRAM = 10
  } axi_slaves_t;

  localparam NB_PERIPHERALS = BRAM + 1;

  localparam logic [63: 0] BRAMLength = 64'h8000;
  localparam logic [63: 0] DebugLength = 64'h1000;
  localparam logic [63: 0] ROMLength = 64'h10000;
  localparam logic [63: 0] CLINTLength = 64'hC0000;
  localparam logic [63: 0] PLICLength = 64'h3FF_FFFF;
  localparam logic [63: 0] UARTLength = 64'h1000;
  localparam logic [63: 0] TimerLength = 64'h1000;
  localparam logic [63: 0] SPILength = 64'h800000;
  localparam logic [63: 0] EthernetLength = 64'h10000;
  localparam logic [63: 0] GPIOLength = 64'h1000;
  localparam logic [63: 0] DRAMLength = 64'h40000000; // 1GByte of DDR (split between two chips on Genesys2)
  localparam logic [63: 0] SRAMLength = 64'h1800000; // 24 MByte of SRAM
  // Instantiate AXI protocol checkers
  localparam bit GenProtocolChecker = 1'b0;

  typedef enum logic [63: 0] {
    DebugBase = 64'h0000_0000,
    ROMBase = 64'h0001_0000,
    CLINTBase = 64'h0200_0000,
    PLICBase = 64'h0C00_0000,
    UARTBase = 64'h1000_0000,
    TimerBase = 64'h1800_0000,
    SPIBase = 64'h2000_0000,
    EthernetBase = 64'h3000_0000,
    GPIOBase = 64'h4000_0000,
    BRAMBase = 64'h5000_0000,
    DRAMBase = 64'h8000_0000
  } soc_bus_start_t;

```
Please, I would like to know if there is another parameter that I forgot to modify?

zarubaf, 2021-08-09:
Did you check the ID width as well, I think people were running over that issue before: https://github.com/openhwgroup/cva6/issues/568 Does that, by any chance solve your issue?

Bill94l, 2021-08-09:
Thank you for your instant response, but this does not solve my problem because the identifier ID changes in the case of adding a master on the bus **[IdWidthSlave = IdWidth + $clog2(NrSlaves)]**.
For example:
```C
package ariane_soc;
  // M-Mode Hart, S-Mode Hart
  localparam int unsigned NumTargets = 3; // debug, ariane, NEW-MASTER
  // Uart, SPI, Ethernet, reserved
  localparam int unsigned NumSources = 30;
  localparam int unsigned MaxPriority = 7;

  localparam NrSlaves = 3; // actually masters(debug, ariane, NEW-MASTER), but slaves on the crossbar

  // 4 is recommended by AXI standard, so lets stick to it, do not change
  localparam IdWidth   = 4;
  localparam IdWidthSlave = IdWidth + $clog2(NrSlaves);
```
My problem at the moment is as soon as I increment the NB_PERIPHERALS parameter in order to add a new SLAVE entry on the bus, nothing works.

Here is the use of this prameter in architecture:
**fpga / src / ariane_xilinx.sv** 
```C
AXI_BUS #(
    .AXI_ADDR_WIDTH ( AxiAddrWidth     ),
    .AXI_DATA_WIDTH ( AxiDataWidth     ),
    .AXI_ID_WIDTH   ( AxiIdWidthSlaves ),
    .AXI_USER_WIDTH ( AxiUserWidth     )
) master[ariane_soc::NB_PERIPHERALS-1:0]();
```

Moschn, 2021-08-09:
Do you have the correct order in the x-bar instantiation? See https://github.com/openhwgroup/cva6/issues/422#issuecomment-866847411

Bill94l, 2021-08-09:
Thank you for your prompt and helpful response. As you can see in the code below, I have kept the order of the devices as they are instantiated in the address range of the AXI Crossbar. 
- Are there other parameters to change? For example the parameter "ariane_pkg :: ArianeCfg.NrPMPEntries = 8" should not be changed to add a new entry on the bus? 

**tb/ariane_soc_pkg.sv**
```C
typedef enum int unsigned {
    DRAM     = 0,
    BRAM     = 1,
    GPIO     = 2,
    Ethernet = 3,
    SPI      = 4,
    Timer    = 5,
    UART     = 6,
    PLIC     = 7,
    CLINT    = 8,
    ROM      = 9,
    Debug    = 10
  } axi_slaves_t;

  localparam NB_PERIPHERALS = Debug + 1;

  localparam logic[63:0] BRAMLength    = 64'h8000;
  ..
  ..
  typedef enum logic [63:0] {
    DebugBase    = 64'h0000_0000,
    ROMBase      = 64'h0001_0000,
    CLINTBase    = 64'h0200_0000,
    PLICBase     = 64'h0C00_0000,
    UARTBase     = 64'h1000_0000,
    TimerBase    = 64'h1800_0000,
    SPIBase      = 64'h2000_0000,
    EthernetBase = 64'h3000_0000,
    GPIOBase     = 64'h4000_0000,
    BRAMBase	 	 = 64'h5000_0000,	 
    DRAMBase     = 64'h8000_0000
  } soc_bus_start_t;
```
**fpga/src/ariane_xilinx.sv**
```C
axi_node_wrap_with_slices #(
    // three ports from Ariane (instruction, data and bypass)
    .NB_SLAVE           ( NBSlave                    ),
    .NB_MASTER          ( ariane_soc::NB_PERIPHERALS ),
    .NB_REGION          ( ariane_soc::NrRegion       ),
    .AXI_ADDR_WIDTH     ( AxiAddrWidth               ),
    .AXI_DATA_WIDTH     ( AxiDataWidth               ),
    .AXI_USER_WIDTH     ( AxiUserWidth               ),
    .AXI_ID_WIDTH       ( AxiIdWidthMaster           ),
    .MASTER_SLICE_DEPTH ( 2                          ),
    .SLAVE_SLICE_DEPTH  ( 2                          )
)
i_axi_xbar (
    ..
    ..
    .start_addr_i ({
        ariane_soc::DebugBase,
        ariane_soc::ROMBase,
        ariane_soc::CLINTBase,
        ariane_soc::PLICBase,
        ariane_soc::UARTBase,
        ariane_soc::TimerBase,
        ariane_soc::SPIBase,
        ariane_soc::EthernetBase,
        ariane_soc::GPIOBase,
        ariane_soc::BRAMBase,
        ariane_soc::DRAMBase
    }),
    .end_addr_i   ({
        ariane_soc::DebugBase    + ariane_soc::DebugLength - 1,
        ariane_soc::ROMBase      + ariane_soc::ROMLength - 1,
        ariane_soc::CLINTBase    + ariane_soc::CLINTLength - 1,
        ariane_soc::PLICBase     + ariane_soc::PLICLength - 1,
        ariane_soc::UARTBase     + ariane_soc::UARTLength - 1,
        ariane_soc::TimerBase    + ariane_soc::TimerLength - 1,
        ariane_soc::SPIBase      + ariane_soc::SPILength - 1,
        ariane_soc::EthernetBase + ariane_soc::EthernetLength -1,
        ariane_soc::GPIOBase     + ariane_soc::GPIOLength - 1,
        ariane_soc::BRAMBase     + ariane_soc::BRAMLength - 1,
        ariane_soc::DRAMBase     + ariane_soc::DRAMLength - 1
    }),
   ..
   ..
);
```

farnam16, 2021-08-09:
Dear @Bill94l,
Your last codes with ordered peripherals seem correct, however, the NrTargets = 3, and in the original code it is equal to 2. (but this is no a problem since I suppose NrTarget is not used in the ariane_xilinx.sv. I list what I have done and worked for me here below: 

1) clone a fresh project
2) without touching anything just add the BRAM as you did in tb/ariane_soc_pkg.sv respecting the order (your last code is correct). 
3) be sure your AxiIdWidthMaster is 4. 
4) I see only a Slave unit (e.g., BRAM) on the Xbar, in your codes. And there is no Master (as listed in your ariane_soc_pkg.sv, 
  but in your ariane_xilinx.sv the NrSlave = 3 which is not correct! it should be 2 if the Peripheral list are as you presented here (It includes only BRAM which is an slave). 
 5) Try to regenerate the Xilinx legacy IPs (MIG, Clock Converter, etc, with the ID width of 6).  
 6) dts (device tree) file is important when the OS is booted for the device driver. it should be modified with added BRAM device and make to regenerate the new bootrom SV file. 
 7) The order of the devices on the Xbar in fpga/src/ariane_xilinx.sv where it is instantiated should be the same as the order where it has been defined (enumerated) in the  tb/ariane_soc_pkg.sv .

Bill94l, 2021-08-10:
Dear @farnam16,
Thank you very much for your helpful reply.
Saluti
Billal

closed 2021-08-10 by Bill94l