openhwfoundation/cva6 #712: Issue with Ethernet module when booting with VC707

Iripi97, 2021-08-24:
I am starting with the following
_git clone https://github.com/openhwgroup/cva6
cd cva6
git submodule update --init --recursive_
I then make sure RISCV is pointed to either the riscv64-unknown-elf toolchain or the riscv64-unknown-linux-gnu toolchain, and have my Vivado tools ready. After modifying ariane.dts (in cva6/fpga/src/bootrom) to remove the ethernet node I run _make all_ in the bootrom directory but receive the following error:

[image]

I even tried adding #include <string.h> to sd.c because that was the proposed solution I found online, but this doesn't fix the error. Is this reproducible with your system/files?
Any help/guidance on this issue would be greatly appreciated. I believed this was an item that had already been solved but now my implementation is getting stuck again at the ethernet node upon trying to boot Linux:

[image]

Bill94l, 2021-09-03:
Hello, 

I think the first problem is with your compilation toolchain.
Here are the steps I followed to install the toolchain
## A) RISC-V tool chain setting up

The tool chain is available to this link:  [https://github.com/riscv/riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain)  At first, you have to get the sources of the RISCV GNU toolchain:

```bash
$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain 
$ cd riscv-gnu-toolchain 
$ git checkout ed53ae7a71dfc6df1940c2255aea5bf542a9c422

```

Next, you have to install all standard packages needed to build the toolchain depending on your Linux distribution. Before installing the tool chain, it is important to define the environment variable `RISCV=”path where the tool chain will be installed”`. Then, you have to set up the compiler by running the following command:

```bash
$ export RISCV=/path/to/install/riscv/compilators
$ ./configure --prefix=$RISCV --disable-linux --with-cmodel=medany --with-arch=rv64imac
$ make newlib 

```

When the installation is achieved, do not forget to add $RISCV/bin to your PATH.

```bash
$ export PATH=$PATH:$RISCV/bin
```

## B) Linux boot problem due to ethernet
You can keep the default device tree with the ethernet node, but you have to generate a new linux image from [ariane-sdk] (https://github.com/pulp-platform/ariane-sdk) toolchain including which you deactivate the ethernet in the configuration of your new linux image.

### Ethernet deactivation:
- In: [ariane-sdk / configs / linux_defconfig] (https://github.com/pulp-platform/ariane-sdk/blob/master/configs/linux_defconfig)

- Look for (ctrl + f) "NET" and deactivate the "NET" configuration, as you can see below:
```bash
# CONFIG_COMPACTION is not set
CONFIG_NET=N
CONFIG_PACKET=y
CONFIG_UNIX=y
CONFIG_TLS=y
CONFIG_INET=N
CONFIG_IP_MULTICAST=y
# CONFIG_WIRELESS is not set
CONFIG_DEVTMPFS=y
CONFIG_DEVTMPFS_MOUNT=y
CONFIG_VIRTIO_BLK=y
CONFIG_EEPROM_AT24=y
CONFIG_NETDEVICES=N
```
- Then run the following command:
```
$ make bbl.bin # generate the entire bootable image
```
After deactivation, linux no longer looks for ethernet at boot time

Good luck
Billal IGHILAHRIZ

Iripi97, 2021-09-13:
@Bill94l Thank you very much for your response. Using the methods you described has worked to resolve the issue.
However if anyone is using a similar process to what I was, another work around is to add _-fno-builtin_ to the CFLAGS definition found in the Makefile (just as it is for the 32-bit target) in **fpga/src/bootrom**. Hope this is helpful.

closed 2021-09-13 by Iripi97